plerup/espsoftwareserial #3: Inverted bits missing

TeaZee, 2015-11-01:
Hi there,

Love your work, but can't use your softserial library without an inverter, because galvanic isolators sometimes invert the 1's and 0's.

The Arduino SoftSerial in the core library supports this. Could you implement this as well?
Great advantage of soft serial over UART is the ability to support this by interpreting inverted.

Please add a boolean with a default false in the constructor so inverted logic can be selected at time of creation.

Kind regards,
TeaZee

luisesn, 2015-11-23:
I don't know much about github but I think I have created a pull request with the changes required to have inverse logic with this library.

In any case this is my fork with it working: https://github.com/luisesn/espsoftwareserial/

plerup, 2015-12-15:
Available now

closed 2015-12-15 by plerup