plerup/espsoftwareserial #8: SoftwareSerial buffer > 255 corrupts data

supersjimmie, 2015-12-17:
I am sorry, I placed this before here:
https://github.com/esp8266/Arduino/issues/1234
But I now think that was not the right place?

I have a serial device that sends roughly 1200 bytes every 10 sec.
During waiting for that, other code needs to keep running, so I need to buffer it all and process it in one call. I then check every 15 sec to fetch all buffered data from the serial buffer.

Example with just a delay to fill the buffer, based on the SoftwareSerial example file:
#include

SoftwareSerial swSer(14, 12, true, 250);

void setup() {
Serial.begin(115200);
swSer.begin(115200);
Serial.println("\nSoftware serial test started");

for (char ch = ' '; ch <= 'z'; ch++) {
swSer.write(ch);
}
swSer.println("");

}

void loop() {
delay(15000);
while (swSer.available() > 0) {
Serial.write(swSer.read());
}
while (Serial.available() > 0) {
swSer.write(Serial.read());
}
}

As long as I use SoftwareSerial swSer(14, 12, true, 250); all goes fine but I lose a lot of data.
(yes, I use inverted = true. that works fine, thank you for that!)
0-1:24.2.1(151217110000W)(00700.778*m3)
0-1:24.4.0(1)
!12C9

When going to a buffer of 260 or higher (need about 1200), data gets corrupted.
¡48224.2.1(151217110000W)(00700.778*m3)
0-1:24.4.0(1)
H&T¦Hø
E72

This is not just because of bad connection. I have tested many times with higher and lower.
Without the delay, nearly no corruption at all. So my data transfers are good.

plerup, 2015-12-21:
The hardware serial implementation (Serial in this case) uses a 256 bytes buffer

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/HardwareSerial.h

closed 2016-02-09 by plerup

thomasvnl, 2017-01-03:
@supersjimmie Did you solve this in the end. I see you're working on DSMR data and I have the same problem. I switched to using the builtin serial because SoftwareSerial doens't seem to handle these speeds very well in conjunction with the ESP8266 (probably because of interrupt handling speed).