plerup/espsoftwareserial #9: 1200 Baud returns null bytes

sk1tt1sh, 2015-12-24:
Not sure if is an issue or I'm doing something wrong, but I'm attempting to read data off a very slow device (1200baud) and am only receiving null bytes, though it is the correct amount.

I verified the data plugging the line and ground from the device into my FTDI and setting the baud to 1200 and received expected values.

SoftwareSerial swSer(13,SW_SERIAL_UNUSED_PIN); //tried setting to 13,12 as well. Only have RX from device though.
void setup() {
  Serial.begin(1200); //according to some posts mixed baud can cause oddities.
  swSer.begin(1200);
  pinMode(ESP8266_LED, OUTPUT);
  delay(100);
}

void loop() {
  while(swSer.available()){
    Serial.print(swSer.read()); //null bytes (the correct amount though)
  }
  delay(100); //raising and lowering doesn't seem to help
}

sk1tt1sh, 2015-12-29:
I flippy flopped the setup to use HW serial for the slow data and sw serial for everything else. As a note, I have a bit of gunk that it writes to the port when it first starts up. So I started sending a long string of A's at startup and giving about 500-700ms delay between swSer.begin(9600); and actually writing data to it. If I do swSer.begin(9600); delay about 100ms and swSer.write("something important"); it tends to spit out shifted bytes.

plerup, 2015-12-30:
Yes, this is a bug. With the new high resolution timing introduced in last commit there is a 16 bit truncation which affects this low baudrate. Fix is on the way

plerup, 2015-12-30:
Fix is in the latest commit. Tested with baud rates down to 300

sk1tt1sh, 2015-12-30:
Awesome! You rock man. I was scratching my head trying to figure out how to make pull request for it... maybe next time :)

closed 2016-01-03 by plerup