plerup/espsoftwareserial #10: Cannot get it to work at < 14400 baud

gubbyhacker, 2015-12-25:
I have an old device that uses 300 baud and inverted logic on its serial interface. I would like to network it. The default Arduino SoftwareSerial for an UNO works fine at 300 baud. Using this library on my Sparkfun Thing dev board does not seem to work at anything less than 14400 baud. I a testing it with an OSEP FTDI board just connecting ground and TX/RX pins. This works at higher baud rates. Just nothing below 14400.

Have you tested this library at lower speeds? (I guess I'm just looking for hope.) Thanks.

sk1tt1sh, 2015-12-28:
If you compare https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/SoftwareSerial/SoftwareSerial.cpp Line 304+ you will see how they set the speed in the arduino SWSer library is significantly different. I'm wondering if he just throw something in to get it at least going. It's also visible in the http://arduiniana.org/libraries/newsoftserial/ library that the speed calc is pretty basic in this version. I'm going to tinker but for now I've switched to using the HW serial for the slow device and SW serial for devices that can move faster/debugging.

plerup, 2015-12-30:
Please try the latest commit, should be fixed now.

closed 2016-02-13 by plerup