plerup/espsoftwareserial #13: SoftwareSerial Rx data error

lable, 2016-01-04:
SoftwareSerial Rx data error.

 Code:

#include <SoftwareSerial.h>

//SoftwareSerial swSer(14, 12, 128);
SoftwareSerial swSer(12, 13, 128);

void setup() {
  Serial.begin(115200);
  swSer.begin(115200);
  //Serial.begin(9600);
  //swSer.begin(9600);

  Serial.println("\nSoftware serial test started");

  for (char ch = ' '; ch <= 'z'; ch++) {
    swSer.write(ch);
  }
  swSer.println("");

}

void loop() {
  if (swSer.available()) {
    Serial.write(swSer.read());
  }
  if (Serial.available()) {
    swSer.write(Serial.read());
  }

}

 tx  "ee", rx "de"

see picture:
[image]

plerup, 2016-01-04:
Are you using the latest version? In that case the parameters for the initiation is wrong. The third parameter is a boolean now and the forth parameter is the buffer size.

SoftwareSerial swSer(12, 13, 128); should be SoftwareSerial swSer(12, 13, false, 128);

lable, 2016-01-05:
Resolved, it is due to the problems caused by the default library reference:
AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\SoftwareSerial
Thank you very much!

closed 2016-01-09 by plerup