plerup/espsoftwareserial #14: Multiple soft serial port does not support?

lable, 2016-01-05:
CODE:
#include <SoftwareSerial.h>

//SoftwareSerial swSer(14, 12, 128);
SoftwareSerial swSer(12, 13,false, 128);
SoftwareSerial swSer2(14, 16,false, 128);

void setup() {
  Serial.begin(115200);
  swSer.begin(115200);
  swSer2.begin(115200);
  //Serial.begin(9600);
  //swSer.begin(9600);

  Serial.println("\nSoftware serial test started");

  for (char ch = ' '; ch <= 'z'; ch++) {
    swSer.write(ch);
  }
  swSer.println("");

}

void loop() {
  if (swSer.available()) {
    Serial.write(swSer.read());
  }
  if (Serial.available()) {
    int buf = Serial.read();
    swSer.write(buf);
    swSer2.write(buf);
  }
  if (swSer2.available()) {
    Serial.write(swSer2.read());
  }
}

OUTPUT:
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

sk1tt1sh, 2016-01-06:
You have your GPIO0 and DTR connected. Try using "RealTerm" and clicking "CLEAR" on DTR for the Pins Tab. Then on the Port tab click open. Then start the esp module.
If it's still not working try printing something out to regular serial before initializing the SwSer instances, initialize one, then print something across it to regular serial, then initialize the second. You should also call yeild() at some point in your loop().

plerup, 2016-01-09:
This was a bug caused by the way the SDK interrupt callback is implemented. Fixed in the latest commit.

lable, 2016-01-17:
Thank you to repair the problem in time！

closed 2016-02-09 by plerup

crosofg, 2017-04-25:
so if I have to use 4 Software SErial ports, which all GPIOs can I use?