plerup/espsoftwareserial #19: Set data, parity and stop bits

WenNanCheng, 2016-03-15:
I try to acquire signals from a device with data bit=7, parity=Even, and stop bit=2.

Is there a way to configure these parameters?

plerup, 2016-03-15:
No, these parameters are currently not implemented. Only 8-bit, no parity and one stop bit is available.

WenNanCheng, 2016-03-15:
I needs to communicate with totally four devices.

This means that I am forced to go back to arduino board. 

Thanks for your help.

Teddyz, 2016-05-03:
I just want to mention that I would love being able to use 9N1 at 19200. In my application first/MSB tell if the rest 8 bits is a command or data. Today this forces me to use a MEGA2560 for a quite stupid task and then send the result via ESP82666 using AT-commands.
How difficult do you think it would be to hack espsoftwareserial to go 9-bit? Any hint is appreciated!

Deggle, 2016-06-25:
Hi guys - I needed to use an even parity, and I thought you may be interested to know that it was relatively simple to update the code to enable this (I was surprised I managed to do it so easily, and it worked on first compile)!

Oh, thought should note that I only implemented this for TX and not for RX but I'm sure that wouldn't be too complex to do if you need it (I'm lazy and this is not critical for me)!

I updated 'begin' to accept a couple of arguments:

`void begin(long speed, int stopbits = 1, char parity = 'N');`

I then created a local variable to hold these arguments (in SoftwareSerial.h), and tweaked the function in SoftwareSerial.cpp to accept and set the variables when 'begin' is called.

Next is to tweak the code for sending the bits - which is all under the 'write' function...

The stop bits was easy to implement - it just needed the WAIT command moving to a loop that calls WAIT for each stop bit needed:

`for (int i = 0; i < _stopbits; i++) { WAIT; }`

Parity was a little trickier, and my implementation may not be the best (most efficient), but it's worked fine for me (at 19200). I found the loop that sends the 8 bits, and before this created a bool to hold the parity bit. It's starting state depends on odd/even parity!

`bool parityBit = _parity!='E' ? false : true;`

Then, for each bit that is output in the existing loop, I flip the parity bit if it's a 1. I've used an XOR to do this as I think this is better than some if statements!?

`parityBit = (b & 1) ^ parityBit;`

Then, once we're done writing the 8 bits (and assuming parity isn't set to 'N') we can output our parity bit as the 9th bit:

`   if (_parity!='N') {
     digitalWrite(m_txPin, parityBit ? HIGH : LOW);
     WAIT;
   }`

I've attached my SoftwareSerial.h and SoftwareSerial.cpp: [SoftwareSerialWithParity.zip](https://github.com/plerup/espsoftwareserial/files/333173/SoftwareSerialWithParity.zip)

I don't often contribute on here, but this seemed like it may be useful for someone?!?

Cheers,

Tim

Dosh22, 2017-12-13:
please add this to master, i need comunicate with 7E1 devices...

closed 2019-03-23 by dok-net