plerup/espsoftwareserial #23: SoftwareSerial avr vs SoftwareSerial esp incompatibility

dextercapo, 2016-07-10:
I ran into a problem , there's a simple communications program using the gt511 fingerprint sensor which works on avr , but I'd like to use an ESP8266 to communicate to it and the basic test programs can't even compile . is there an incompatibility in calls between the two ?
here's the lib for the gt511:
http://wordpress.hawleyhosting.com/ramblings/Downloads/FPS_GT511C3.zip
http://wordpress.hawleyhosting.com/ramblings/?p=375
the error codes I receive are : "error: SoftwareSerial has no member named 'end' , 'listen'
thanks in advance

plerup, 2016-07-25:
Well, in espsoftwareserial all ports are in listening mode all the time so there was no need for this function. I guess a dummy one could be added for compatibility reasons.

dextercapo, 2016-07-25:
thanks , I thought the same , but I'm not too fluent in arduino yet, soI'm still poking around to make the blink test routine work . 
thanks for the reply.

danielcbit, 2016-08-17:
And what about the end function? Can it be implemented too?

dextercapo, 2016-08-17:
None of them was needed. An empty function did the trick.
Now it works fine.
Bye
dexter

plerup, 2016-08-21:
Included in latest commit

closed 2016-11-06 by plerup