plerup/espsoftwareserial #25: Data Corruption

PhilColbert, 2016-08-13:
Hi

Firstly thanks for the contribution, great effort! 

I have a small problem that theres quite a bit of data corruption coming in from a GPS device.

Have played with the buffer size and trying flush before read or write but still the problem remains?

Seems adding a delay around the read and write time definately helps but doesnt clear the problem completely.

Any ideas, I have this one problem left in a big project, so close but so far ! :)

Thanks for any help !

PhilColbert, 2016-08-13:
Weirdly enough, caused by  this line

free_mem = system_get_free_heap_size();

Removed that call and all corruption gone, how weird, just in case someone has this problem in future. 

Thanks

Raphyyy, 2016-11-21:
Hi,
Was this line in your code or in core ?
I have corrupted data too while communicating with a SIM808 (GSM+GPS module) and I'm unable to fix it...

closed 2019-03-23 by dok-net