plerup/espsoftwareserial #26: WDT Reset when using mDNS and ESP8266WebServer

liquidfalcon, 2016-08-18:
So I'm seeing the following wdt reset very occasionally (3-5 times in a 12 hour period) when using both espsoftwareserial, mDNS, and ESP8266WebServer together continuously:

`dev 1143

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v5b34dc19
~ld
`

Disabling espsoftwareserial allows the program to run fine indefinitely. Now, the only thing helpful here seems to be the rather obscure "dev 1143" line, which leads to a single forum post on Espressif's site here: http://bbs.espressif.com/viewtopic.php?t=1205 - But, the OP states he could reproduce it via os_intr_lock/unlock - I tried disabling the `sei()` / `cli()` calls, but, that made no difference, so I'm rather stumped on what else it could be other than the receive interrupt being slow. 

Now, this wdt reset _always_ happens when I'm waiting for a response on the software serial line. Immediately before this I would have answered and completed an HTTP response - But, I do not know if they're necessarily related.

WheresWally, 2016-10-29:
I am getting 1 to 4 WDT reset every hour, my sketch is reading 200-300 bytes @ 9600 in a block every 6 seconds. Every minute the sketch uses WiFi.client to send data to various servers and devices.

Using a NodeMCU1.0 Arduino 1.6.12 ESP 2.3.0 and softwareserial RX on pin 13.

Makuna, 2016-11-08:
Definitely the disabling interrupts for any length greater than 10us will be an issue for core WiFi support and generally I have found need to be avoided on esp8266.
Its a shame as I really need an extra UART in my project.

riegaz, 2017-04-20:
Any solution for this problem so far? Facing the same with SPI and TCP client.

closed 2019-03-23 by dok-net