plerup/espsoftwareserial #28: GPIO0, GPIO2, and GPIO15 are boot select pins in Arduino style

sk1tt1sh, 2016-08-29:
Hello,

I have determined an inconsistent boot cause to be due to the way the way this library configures pins for interrupt. This is especially apparent when using the built-in-led on gpio2 of the ESP-12e board.

```
#define BUILT_IN_LED 2

SoftwareSerial softs(4,5,false,128);

void setup(){
  softs.begin(38400);  
  pinMode(BUILT_IN_LED, OUTPUT); <<-- Crash. Module will then go into UART boot mode as 0, and 2 will go high
}
```

These pins go high when `void ICACHE_RAM_ATTR sws_isr_0() { ObjList[0]->rxRead(); };` is called.

-sk1

plerup, 2016-08-29:
I don't get this. You are using pin 4 and 5 for software serial. Where does GPIO 0 come into this?

sk1tt1sh, 2016-09-12:
Thanks, I don't entirely understand it either. I will attempt to create some more detail with debugging when I get a chance. I have just commented out 0,1 and 15 "ICACHE_RAM_ATTR" functions and the corresponding lines in the ISRList are NULL and it has resumed working. (Pin 1 is the built-in TX pin on the esp-12e)

I have confirmed with an adafruit huzzah.
```
#define RXS 4
#define TXS 5

SoftwareSerial ssp(RXS,TXS,false,128);

void setup(){
  Serial.begin(38400);
  delay(100);
  ssp.begin(38400);
  delay(100);

  Serial.println("Started");
  ssp.write("Test!");
}

void main(){
}
```

So to test this program the module, unplug it entirely, then with  your FTDI connected to the SoftwareSerial pins and nothing on the hardware serial. The device should blink the built in LED when you power it but will never actually finish booting. If you connect to the hardware serial it will inconsistently boot.

Thanks!

closed 2019-03-23 by dok-net