plerup/espsoftwareserial #32: Frequent WDT resets

dotJson, 2016-10-27:
Console shows periodic WDT resets.  SoftwareSerial intialized as 12,14,false,128.  I'm using the newest library from this repos. 

The software serial connects at 9600.  I cannot find the definition for rst cause code 4 (so far).
The console-to-onboard serial is 115.2k.

WDT output
```
ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

WheresWally, 2016-10-29:
[https://github.com/esp8266/Arduino/pull/1018/commits/9db650482cc255e4d49fb3e9a8bf6ad0a5fdc900](url)

```
 ### Improved Stability
  ![ESP improved stability](ESP_improved_stability.png)
 +
 +### Boot Messages and Modes
 +
 +The ESP module checks at every boot the Pins 0, 2 and 15.
 +based on them its boots in different modes:
 +
 +| GPIO15 | GPIO0 | GPIO2 | Mode                             |
 +| ------ | ----- | ----- | -------------------------------- |
 +| 0V     | 0V    | 3.3V  | Uart Bootloader                  |
 +| 0V     | 3.3V  | 3.3V  | Boot sketch (SPI flash)          |
 +| 3.3V   | x     | x     | SDIO mode (not used for Arduino) |
 +
 +
 +at startup the ESP prints out the current boot mode example:
 +```
 +rst cause:2, boot mode:(3,6)
 +```
 +
 +#### rst cause
 +
 +| Number | Description            |
 +| ------ | ---------------------- |
 +| 0      | unknown                |
 +| 1      | normal boot            |
 +| 2      | reset pin              |
 +| 3      | software reset         |
 +| 4      | watchdog reset         | 
 +
 +
 +#### boot mode
 +
 +the first value respects the pin setup of the Pins 0, 2 and 15.
 +
 +| Number | GPIO15 | GPIO0 | GPIO2 | Mode       |
 +| ------ | ------ | ----- | ----- | ---------- |
 +| 0      | 0V     | 0V    | 0V    | Not valid  |
 +| 1      | 0V     | 0V    | 3.3V  | Uart       |
 +| 2      | 0V     | 3.3V  | 0V    | Not valid  |
 +| 3      | 0V     | 3.3V  | 3.3V  | Flash      |
 +| 4      | 3.3V   | 0V    | 0V    | SDIO       |
 +| 5      | 3.3V   | 0V    | 3.3V  | SDIO       |
 +| 6      | 3.3V   | 3.3V  | 0V    | SDIO       |
 +| 7      | 3.3V   | 3.3V  | 3.3V  | SDIO       |
 +
 +Note:
 +
 +number = ((GPIO15 << 2) | (GPIO0 << 1) | GPIO2);
 +
 +

beachmiles, 2016-11-07:
I'm seeing lots of soft wdt resets using the latest software with one softserial at 115000 and one at 9600. It looks like during init the softserial at 9600 gets a lot of garbage from the GPS nmea ai thinking A7 module I am turning on.

closed 2019-03-23 by dok-net