plerup/espsoftwareserial #42: EspSoftwareSerial stop wifi

blacktrooper2, 2017-01-29:
Hi there,

I'm programming an Esp8266 from Espressif (vroom-02).
The project is to sense with an Arduino Uno and then send data to Esp8266 which will send to computer over wifi.

The wifi alone is working, the softwareserial alone as well.
But the combination of both stops wifi.
I think it is because fo interrupts attached to Rx pin for communication communication.

So I tried to use stopListening but I didn't solve my issue.

Have an idea ?

Chris

blacktrooper2, 2017-01-29:
The ESP 8266 side code:

`

// ESP serial pin to use shall be IO2, IO4 or IO5 which supports interrupt
// Reverse pin : Tx ESP <-> Rx Arduino

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
const char*         ssid                = "Z3K";                    // MOD
const char*         password            = "iloveyou";               // MOD
#define             sendIP                "192.168.43.221"          // MOD
unsigned int        sendPort            = 8889;                     // MOD
#define             tempoUs               500                       // MOD
boolean             wifiConnected       = false;
boolean             udpConnected        = false;
char                wifiData[255];
WiFiUDP Udp;

#include <SoftwareSerial.h>
#define             RXpin                 2                         // MOD
#define             TXpin                 5                         // MOD
SoftwareSerial espSerial(RXpin, TXpin);
boolean             espSerialNewData;
char                espSerialBuffer;
String              serialData;

unsigned long       counter             = 0;                        // suppr

// Wifi et Signal drop LED
#define connectionLEDpin 12
#define dropLEDpin 13

void setup() { // ----------------------------------------------------------------------
  pinMode(connectionLEDpin, OUTPUT);
  pinMode(dropLEDpin, OUTPUT);
  digitalWrite(connectionLEDpin, LOW);
  digitalWrite(dropLEDpin, LOW);

  wifiConnected = connectWifi();
  if (wifiConnected) {
    udpConnected = connectUDP();
  }
  wifiDetails();
  ESP.wdtDisable();
  ESP.wdtEnable(WDTO_8S);

  Serial.begin(115200);
  Serial.println(F("\nHardware serial initialized"));
  delay(10);
  espSerial.begin(57600);
  Serial.println(F("Software serial initialized"));
  delay(10);
}

void loop() { // -----------------------------------------------------------------------
  testBoucle();
  //  GetEspSerial　Data();
}

void StringToTable() { // --------------------------------------------------------------
  strcpy(wifiData, serialData.c_str());
}

void sender() { // ---------------------------------------------------------------------
  //  detachInterrupt(RXpin);
  //  detachInterrupt(TXpin);
  espSerial.stopListening();
  checkWifiStatus();
  StringToTable();
  Udp.beginPacket(sendIP, sendPort);
  Udp.write(wifiData);
  boolean state = Udp.endPacket();
  if (!state) {
    Serial.print(F("Error during sending: "));
    Serial.print(wifiData);
    Serial.println(counter);
    digitalWrite(dropLEDpin, HIGH);
  }
  ESP.wdtFeed();
  yield();
  delayMicroseconds(tempoUs);
  espSerial.listen();
  //  attachInterrupt(RXpin,HIGH);
  //  attachInterrupt(TXpin,HIGH);
}

boolean connectWifi() { // -------------------------------------------------------------
  boolean state = true;
  int i = 0;
  WiFi.begin(ssid, password);
  Serial.print(F("\n\n\n"));
  Serial.print(F("\n***************************************************"));
  Serial.print(F("\n...Connecting WiFi..."));
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    if (i > 10) {
      state = false;
      break;
    }
    i++;
  }
  if (!state) {
    Serial.print(F("Wifi Error"));
  }
  else {
    Serial.print(F("Wifi Successful"));
  }
  return state;
}

boolean connectUDP() { // --------------------------------------------------------------
  boolean state = true;
  int i = 0;
  Serial.print(F("\n...Connecting  UDP..."));
  while (Udp.begin(sendPort) != 1) {
    //    delay(500);
    if (i > 3) {
      state = false;
      break;
    }
    i++;
  }
  if (!state) {
    Serial.print(F("UDP Error"));
  }
  else {
    Serial.print(F("UDP Successful"));
  }
  return state;
}

void checkWifiStatus() { // ------------------------------------------------------------
  if (WiFi.status() == WL_DISCONNECTED) {
    Serial.println(F("Wifi disconnected"));
    digitalWrite(connectionLEDpin, LOW);
  }
  else {
    digitalWrite(connectionLEDpin, HIGH);
  }
}

void wifiDetails() { // ----------------------------------------------------------------
  Serial.println(F("\n***************************************************"));
  Serial.print(F("SSID:                       "));
  Serial.println(ssid);
  Serial.print(F("PWD:                        "));
  Serial.println(password);
  Serial.print(F("ESP8266 IP address:         "));
  Serial.println(WiFi.localIP());
  Serial.print(F("Computer IP address:        "));
  Serial.println(sendIP);
  Serial.print(F("Send port:                  "));
  Serial.println(sendPort);
  Serial.print(F("Signal:                     "));
  Serial.print(WiFi.RSSI());
  Serial.println(F(" dBm"));
  Serial.println(F("***************************************************"));
}

void GetEspSerialData() { // ----------------------------------------------------------------
  counter = counter + 1;
  espSerialNewData = false;
  char espSerialBuffer;
  String serialData;

  espSerial.print(1);
  while (espSerial.available() > 0) {
    espSerialNewData = true;
    espSerialBuffer = espSerial.read();
    serialData.concat(espSerialBuffer);
  }

  if (serialData != "") {
    sender();
    Serial.print(F("sent wifi:"));
    Serial.println(serialData);
  }
  //  else {
  //    Serial.print(F("soft serial empty "));
  //    Serial.println(counter);
  //    serialData = "Loop " + String(counter) + " empty";
  //    sender();
  //  }
}

void testBoucle() { // -----------------------------------------------------------------
  for (int i = 0; i < 1024; i++) {
    serialData = i;
    sender();
  }
  for (int i = 1024; i > 0; i--) {
    serialData = i;
    sender();
  }
}

`

blacktrooper2, 2017-01-29:
Arduino side:
``// Mega 2560 Rx: 10, 11, 12, 13, 50, 51, 52, 53, 62, 63, 64, 65, 66, 67, 68, 69
// Reverse pin : Tx ESP <-> Rx Arduino

#include <SoftwareSerial.h>
#define     RXpin      10                     // MOD
#define     TXpin      11                     // MOD
String      Data;
SoftwareSerial espSerial(RXpin, TXpin);

void setup() { // -----------------------------------------------------------------------
  Serial.begin(115200);
  Serial.println(F("\nHardware serial initialized"));
  espSerial.begin(57600);
  Serial.println(F("Software serial initialized"));
}

void loop() { // ------------------------------------------------------------------------
  for (int i = 0 ; i < 10; i++) {
    while (espSerial.available() == 0) {}
    espSerial.read();
    espSerial.print(String(i));
    Serial.print("soft sent: ");
    Serial.println(i);
  }

  for (int i = 10 ; i == 0; i--) {
    while (espSerial.available() == 0) {}
    espSerial.read();
    espSerial.print(String(i));
    Serial.print("soft sent: ");
    Serial.println(i);
  }

  Serial.println();
}``

reischle, 2017-03-21:
Same issue here. Have a continuous stream of data coming in at 2400 baud. RX-only.
WiFi just dies after a few seconds.

bestpika, 2018-07-06:
I got same issue...

closed 2019-03-23 by dok-net