plerup/espsoftwareserial #51: Possible bug detected

PedroD, 2017-06-17:
In the `WAIT` macro, what happens if the `` overflows and starts over from 0?

line: https://github.com/plerup/espsoftwareserial/blob/master/SoftwareSerial.cpp#L156

`#define WAIT { while (ESP.getCycleCount()-start < wait); wait += m_bitTime; }`

Imagine that `ESP.getCycleCount()` overflowed and is now much much lower than `start`. The wait will take lots of time to recover, and the following `wait` values will be wrong.

PedroD, 2017-06-17:
Actually, in practice this "bug" is negligible for very small intervals (m_bitTime), so no harm done!

plerup, 2017-10-17:
This is unsigned 32 bit arithmetic so this works also around possible overflow

closed 2017-10-17 by plerup