plerup/espsoftwareserial #54: Exceptions with baudrate 9600

UlliBien, 2017-12-03:
Hi,
I'm using espsoftwareserial to communicate with a PZEM-004T using library [https://github.com/olehs/PZEM004T](url).

I got a lot of reading errors and mostly an exceptions occured after a few seconds. Somtimes it was stable for nearly a minute, but never longer than two minutes. When I got an exception epc1 mostly pointed to `optimistic_yield()`.

In the interrupt-handler rxRead I found the line `WAIT;`. This macro expands to
`#define WAIT { while (ESP.getCycleCount()-start < wait) if (!m_highSpeed) optimistic_yield(1); wait += m_bitTime; }`

`m_highSpeed` is set in `begin()`: `m_highSpeed = speed > 9600;`. The baudrate for the PZEM-004T is 9600 bps. So `m_highSpeed` is set to false and `optimistic_yield()` is called.

I set `m_highSpeed` to `true` for my special case. All the communication errors disappeared. And now more exceptions occured. Currently my software is running stable for more than a hour.

Yous should revise the definition of `WAIT`. In the code for `WAIT` in the original library is `#define WAIT { while (ESP.getCycleCount()-start < wait); wait += m_bitTime; }`

Kind regards
UlliBien

TD-er, 2017-12-09:
Indeed, using 9600 baud will result in timeouts and corrupted messages.
Simple change in begin():
old:   `m_highSpeed = speed > 9600; `
new:   `m_highSpeed = speed >= 9600;`
Results in stable reading.

@UlliBien What do you mean with "more exceptions"?

UlliBien, 2017-12-10:
Sorry, 
not "now more", "no more" instead. After making sure m_highSpeed was true, the exceptions disappeared.

devyte, 2018-02-06:
@TD-er said:

> Simple change in begin():
> old: m_highSpeed = speed > 9600;
> new: m_highSpeed = speed >= 9600;

@plerup there is one user in the Arduino repo who confirms the proposed fix. Without it, behavior is flaky.

plerup, 2018-02-06:
As your mileage may vary here I'm going to add a new method where you can specify what "highSpeed" actually is, but the default will be changed as you suggest

plerup, 2018-02-26:
Done in latest commit

closed 2019-03-23 by dok-net