plerup/espsoftwareserial #55: SoftwareSerial causing spurious resets

eykamp, 2018-01-20:
I'm having a problem with SofwareSerial causing resets on my ESP8266 (NodeMcu 12E).  I have a sensor which sends data via serial connection, which I'm reading using SoftwareSerial on pins D7 and D8, at 9600bps, with a 1024 byte buffer (I've experimented with different buffer sizes with no apparent change of behavior).  Typically, the device will run for 30-90 minutes, and then reset due to a triggering of the hardware watchdog.  It will sometimes reset sooner (10 minutes), but I've never seen it live longer than 2 hours before resetting.

The data being sent by the sensor is small packets (< 512 bytes), spaced out about a second or two apart.

If I disconnect the serial device, the resets go away (probably because there is no data being read).  This would seem to eliminate everything except the code that actually reads the data.

If I leave the device connected, but remove the SoftwareSerial code, the resets go away.  This would seem to eliminate electrical faults or overloading, or similar problems.

I'm working with the current version, but have tried older versions as well (2.0.0 and 2.1.1) with no change in behavior.

My latest experiment involves removing the WAIT code in SoftwareSerial::rxRead().  This, of course, makes the data read garbage, but it also cures the crashes.

My hyphothesis is that something in the WAIT function is causing the device to go into a long loop (or lock) such that the hardware watchdog steps in and reboots the device.

Any ideas?

My code is long and mostly irrelevant, but it is available here:

https://github.com/eykamp/birdhouse/blob/master/firmware/firmware.ino

horendus, 2018-03-07:
I’m also finding this.

I’m using 1 wire, no Rx line in use, and I have used the disable interrupt function on the Rx line added just recently which didn’t clear up the random reboots.

jeroenst, 2018-03-26:
I also can confirm softwareserial is causing my esp to have watchdog timeouts every several hours.

It occurs with read and write of software serial.

logging:
read_MHZ19()
executing co2Serial.write(cmd,9)
dev 1153

ets Jan 8 2013,rst cause:4, boot mode:(1,6)

wdt reset

jeroenst, 2018-03-26:
It look like replacing 
#define WAIT { while (ESP.getCycleCount()-start < wait) if (m_intTxEnabled) optimistic_yield(1); wait += m_bitTime; }

with 
#define WAIT { delayMicroseconds(m_bitTime-1); } // Wait bittime minus some execution time

solves the problem of watchdog timeouts.

Is it possible the calculation fails when the ESP.getCycleCount() returns to 0 after 4294967295 and causing a long delay that triggers the watchdog?

closed 2019-03-23 by dok-net