plerup/espsoftwareserial #56: One Wire Half Duplex Serial communication

Super169, 2018-01-23:
Will you consider to add the support of one wire half duplex serial communication in the SoftwereSerial library?

I have modified the library and tested with the servo of UBTech Alpha 1S robot using one wire half duplex communication in 115200 bps.  (https://github.com/Super169/espsoftwareserial/tree/dev )

One wire communication will be enabled if receivePin == transmitPin, the data pin is set to Rx mode by default, and a new method enableTx has been added set the data pin to Tx mode, it must be called before calling write method.  I have tried to put the enableTx logic inside write method, but it will toggle the interrupt frequently, so I decided to have a separate method, and called before and after the whole transmission.

I'm a newbie in ESP8266 programming, there should have better approach to handle such communication.  It'd be better if the SoftwareSerial library can be enhance to support it officially.

Thanks in advance.
James

closed 2018-02-26 by Super169

PaulWieland, 2022-04-04:
@Super169 were you successful in getting one wire half duplex working? I have a project where I need a one wire uart.

Super169, 2022-04-25:
Yes, it works fine in my RobotControl project using one wire half duplex connection to control the serial servo.

Super169, 2022-04-26:
In my RobotControl or UBTech project, already shared in GitHub.
In fact, the reason I added this feature in software serial was mainly for the RobotControl project, the servo using in this robot is a serial servo from UBT by one wire half duplex serial connection.

dok-net, 2022-04-26:
@PaulWieland Are you aware that EspSoftwareSerial supports one-wire for the last few years? On ESP8266, it's part of the Arduino core releases, on ESP32, use the library manager, or clone from Github.