plerup/espsoftwareserial #60: [BUGFIX] SoftwareSerial locks up when data is present during boot

JaapBraam, 2018-02-07:
If there is data present on an rx pin during startup, the ESP8266 locks up and eventually restarts.

I think this is caused by [https://github.com/plerup/espsoftwareserial/blob/master/SoftwareSerial.cpp#L84](enableRx(true); on line 84). This is called before calling  [https://github.com/plerup/espsoftwareserial/blob/master/SoftwareSerial.cpp#L94](begin(9600); on line 94). 
enableRx(true) starts the interupthandler for the rx pin, but m_bitTime has not been set yet! So m_bitTime will effectively be 0. This causes the WAIT to be very busy, hanging up the system.

If I comment out line 84, my ESP8266 starts up as expected when data is present on the rx pin. enableRx(true) will be called when begin(9600); on line 94 is called, so that doesn't change anything.

plerup, 2018-02-25:
Good find, thanks

plerup, 2018-02-26:
Fixed in latest commit

teetsd, 2018-04-04:
Hi,
I have just run into same issue and will comment line 84 (until next lib update).
Nonetheless, reading enableRx function, I'm also wondering if "m_rxEnabled = on" is in right place (?).  I would place ON 2 lines upper and set this one to OFF. 
Thanks a lot for your comment.

akkaman007, 2018-08-24:
Hi i tried the same thing as suggested by JaapBraam. But still my esp32 is rebooting again and again and showing GURU Medition Error

closed 2019-03-23 by dok-net