plerup/espsoftwareserial #70: Data corruption when send SoftwareSerial data

VisarutR, 2018-04-12:
Hi
I have to create a simple SoftwareSerial test with PIN4 and PIN5 with ESP12F DEVKIT V3
My code is sending out data every 100ms with baud rate 9600, my serial connect to Docklight program.
Data that i want to send is **0x01 0x02 0x03 0x04 0x05 0x06 0x07**, when Docklight receive data program will answer back with **0x0A 0x0B 0x0C**

So the problem is I found a random data corruption when send out data.
Please advice me, if  you have any suggestion.

[image]

[image]

Here is my simple code:
```
#include <SoftwareSerial.h>
#define SSerial2_PINRX      (5)
#define SSerial2_PINTX      (4)
 
SoftwareSerial SSerial2(SSerial2_PINRX, SSerial2_PINTX, false, 256);  // RX, TX, inverse_logic, buffSize  

static uint8_t send_data[20];
static int tmp_size = 0;
static int tx_i;

static uint32_t currentMillis = 0;
static uint32_t previousMillis = 0;
static uint32_t interval = 10;
static uint32_t timer_20ms = 0;
static uint32_t timer_100ms = 0;
static uint8_t timer_20ms_tick = 0;
static uint8_t timer_100ms_tick = 0;

void setup() {
  // init serial
  SSerial2.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:

  currentMillis = millis();
  if(currentMillis - previousMillis > interval) {
    previousMillis = currentMillis;

    timer_20ms++;
    if(timer_20ms >= 2){
      timer_20ms = 0;
      timer_20ms_tick = 1;
    }

    timer_100ms++;
    if(timer_100ms >= 10){
      timer_100ms = 0;
      timer_100ms_tick = 1;
    }
  }

  if(timer_20ms_tick){
    timer_20ms_tick = 0;
  }
  
  if(timer_100ms_tick){
    timer_100ms_tick = 0;
    memset(send_data, 0, sizeof(uint8_t)*20);
    tmp_size = 0;

    send_data[0] = 0x01;
    send_data[1] = 0x02;
    send_data[2] = 0x03;
    send_data[3] = 0x04;
    send_data[4] = 0x05;
    send_data[5] = 0x06;
    send_data[6] = 0x07;

    tmp_size = 7;

    for(tx_i=0; tx_i<tmp_size; tx_i++){
      SSerial2.write(send_data[tx_i]);
    }
  }
}
```

VisarutR, 2018-04-18:
Anyone can help me about this?

Super169, 2018-04-19:
I also encountered problem in 9600bps (with GPIO 14, 13), but it works fine in 115200bps.

I'm still studying if this is related to low speed communication, since the interrupt will be disabled during transmission only if m_highSpeed (m_highSpeed = speed > 9600;).   i.e. the transmission can be interrupt for 9600bps.  However, disable interrupt for low speed communication may have other impact as the transmission time is relatively long.  

I'm not sure if it's really related to the interrupt setting, could you please try if your problem can be solve in other speed above 9600bps?

VisarutR, 2018-04-19:
Hi,
Thank you for your advice. 

So, I have test with same scenario by send out data with baud rate(14400, 19200, 28800, 38400, 56000, 57600 and 115200) every 100ms for 5 minutes, all of them work fine.

That's meaning the problem occurs only 9600bps.

Super169, 2018-04-20:
I just find that it has already updated to disable interrupt in 9600 also.
Please refer to the commit  https://github.com/plerup/espsoftwareserial/commit/7077979af30f12454b8e3165df8f41c73c5a2d24 

Old version:

` m_highSpeed = speed > 9600;`

New version:

`m_intTxEnabled = speed < 9600;`

Have you try the latest version already?

VisarutR, 2018-04-20:
Previously, I installed SoftwareSerial from Arduino library manager, it provided SoftwareSerial v1.0.0 and cannot be upgrade.

I have to pull latest version from this git.
Currently it's working fine more than 10 mins.
I will test over the week and give feedback to you again.

VisarutR, 2018-05-24:
Hi,
I have test over a month with my full program.
Serial communication is looking good.
Once a week, still found some problem but not sure it's related with corruption or not.

closed 2019-03-23 by dok-net