plerup/espsoftwareserial #77: Change recommendations

juancgalvez, 2018-07-21:
Hi Peter.

I have been using you software serial for a long time now. I did some changed I posted in another issue to make it work fine. This was and old Software Serial version.

There are some recommendations I am going to make to avoid issues to users.

Base on github line numbering where doing this post I recommend you to should consider:

**1.** add call to yield() function inside available(), read() and peek() functions. I read [this](https://github.com/plerup/espsoftwareserial/issues/55) post where [jeroenst](https://github.com/jeroenst) wrote something like this:

```
  uint8_t cmd[9] = {0xFF, 0x01, 0x86, 0x00, 0x00, 0x00, 0x00, 0x00, 0x79};
  uint8_t response[9]; // for answer

  co2Serial.write(cmd, 9); //request PPM CO2

  
  // The serial stream can get out of sync. The response starts with 0xff, try to resync.
  while (co2Serial.available() > 0 && (uint8_t)co2Serial.peek() != 0xFF) {
    co2Serial.read();
  }

  co2Serial.readBytes(response, 9);
```
In his while cycle he never yields. If the currently available characted is 0xFF then the program keeps in an endless cycle making (In fact I see this as an error in coding, anyways his program is going to be in an endless loop) wdt resets the device.

**2.** When non inverting, once the transmit pin is enabled, it must be kept HIGH until the start bit is sent, which puts TX pin LOW for m_bitTime. Stop bit must set pin HIGH for m_bitTime and remain HIGH.

When inverting, TX pin must be LOW when TX enabled and set HIGH for start bit; stop bit must set pin LOW for m_bitTime and remain LOW.

Considering above explanation:

- lines 190 is absolutely unnecessary.
- Line 193 should be:

`digitalWrite(m_txPin, m_invert);`

- Line 201 should be:

`digitalWrite(m_txPin, !m_invert);`

**3.** In line 227 where you are doing this:

`unsigned long wait = m_bitTime + m_bitTime/3 - 500;`

Basically you are making longer/shorter the bit time (or wait if you want to see it like because what make the bit duration is wait time). Lets see some examples:

At 115200 bps if we assume 80Mhz clock the m_bitTime  would be 694 (80000000/115200=694 doing integer division). If we apply the code at line 227 we get 694 - 231 - 500 which equals 425  considering unsigned integer. 425 is going to be the wait for every one bit measurement. Now, 80000000/425=188235 that was not the original intended speed.

What you need to do is a delay before reading date bits that put the bit measurement at the middle of a bit. I would recommend you to replace lines 227 and 228:

```
  unsigned long wait = m_bitTime + m_bitTime/3 - 500;
  unsigned long start = ESP.getCycleCount();
```

with:

```
  unsigned long start = ESP.getCycleCount();
  unsigned long wait = m_bitTime/2;
  WAIT;  // delay half a bit
  wait = m_bitTime;
```
 I have not considered the 500 constant that could be necessary to adjust and include.

Probably doing some measurements with a Logic Analyzer could adjust timing.

Respectfully,

Juan C Galvez

mikekgr, 2018-08-17:
Dear @juancgalvez ,
today and because I had some problems, came here to search for a possible solution to my problem that is using ESP8266, Arduino Core 2.4.2 and this library for 9600bps serial communication.  I had problems with wdt restarts etc. All are solved when I did hardware serial connection/communication.
Please tell me if you have somewhere posted the modified file ( I think is SoftwareSerial.cpp the only modified file, am I right? ) and if you did testing to ensure better/stable operation using your changes in the library.

Please try to answer to me in details taking in consideration that I am very amateur programmer... !

Thanks and Best Regards,
Mike Kranidis

mikekgr, 2018-08-18:
Dey @juancgalvez  thanks for your reply. I saw your GitHub but your implementation is not for ESP8266 Arduino core and I cannot convert it. I f you can assist me somehow I can test it to my project. Thanks in advance!

juancgalvez, 2018-08-18:
@mikegr:

That is not my software. I wrote a wrapper and did some modifications to [pleningerweb's](https://github.com/plieningerweb/esp8266-software-uart) software to make it work with Arduino ESP8266.

You can get what I wrote [here](https://github.com/juancgalvez/Arduino-esp8266-Software-UART).

Peter, excuse me for writing this link here. Your software is excellent. I guess most people have issues with it because they don't  do delay() or yield() in their programs and WDT fires.

human890209, 2018-09-03:
@juancgalvez 

>   unsigned long wait = m_bitTime + m_bitTime/3 - 500;
>   unsigned long start = ESP.getCycleCount();
> 
>   unsigned long start = ESP.getCycleCount();
>   unsigned long wait = m_bitTime/2;
>   WAIT;  // delay half a bit
>   wait = m_bitTime;

I think the reason of rxRead()'s WAIT is a linear formula is that Different CPU Frequency will get a different lag of interrupt, Using a fixed offset could get problems under different F_CPU, but I guess using m_bitTime is also not 100% safe, maybe F_CPU is a real factor for this time fix.

juancgalvez, 2018-09-06:
@human890209: You are right. My point 3 is incorrect. I am going to fix it.

I did some measurement with a logic analyzer. Normally, the interrupt lag is 1 to 2 microseconds. Some times it is 12 microseconds (Probably he ESP8266 is serving another interrupt or has disabled interrupts to do something else). At 115200 bps a bit time is 8.68 microseconds so 12 microseconds means the first bit is lost and every other bit is invalid.

human890209, 2018-09-11:
@juancgalvez 
Yes, The interrupt is not 100% stable for ESP8266, even for ESP32 (but it's much better).
The first byte will break the whole line if there is no delay between each byte. And a lower BAUD which makes the bit time longer could improve the stability.
I'm going to add data verification to my ESP32( though I get almost 99.9999% correct content), but in case the cable is disconnected or some physical damage is dealt with the connecting cable. My robot is moving and shaking. The verification is necessary. With verification added, I think ESP8266 could do this job very well, too.

dok-net, 2019-03-23:
Please check if issue still persists in release v4.0.0, otherwise please close it.

closed 2019-04-03 by dok-net