plerup/espsoftwareserial #80: WDT reset on wakeup from deep sleep

Pman123, 2018-09-27:
I have discovered that software serial causes the ESP8266 to undergo a WDT reset on wakeup.  It works fine in the code once the chip re-boots but is adding an unwanted extra 6 or more seconds of high power battery consumption which is most of the point of using deep sleep.

I found that simply adding the line
             SoftwareSerial swSer(rxPin, txPin, false, 128);  //Define sensor connections
without any other lines like 
             swSer.begin(9600); 
is enough to produce the hangup on wake up.

This is the output from the hardware serial pins.

--> Deep sleep
sd.$úü|.Ñl‡|....å.lÏ.c|é√.‰.ì;ìcå.cå˚g'ülnoú„Ï.b.xÑèl;l{lpÚ'‡....É.l.åú...c.o„<.l‰..ècÑ˚gnÁ.dƒád`.ò..gn.d`...oséõìn..c.ú.d.{íõo..b.ú.$ú.‰.l‰ápéÏ«.l¸√'ú.
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0c897c37
~ld 

I assume its something to do with the use of interrupts but have been unable to find a workaround.

Thanks Martin.

dok-net, 2019-03-23:
Can you please verify your reported issue against release v4.0.0?

dok-net, 2019-04-03:
I've patched the StressTest sketch:

diff --git a/libraries/esp8266/examples/SerialStress/SerialStress.ino b/libraries/esp8266/examples/SerialStress/SerialStress.ino
index 12ca68d4..1bbde25b 100644
--- a/libraries/esp8266/examples/SerialStress/SerialStress.ino
+++ b/libraries/esp8266/examples/SerialStress/SerialStress.ino
@@ -63,6 +63,8 @@ void preinit() {
   ESP8266WiFiClass::preinitWiFiOff();
 }
 
+uint32_t deepSleepDeadline;
+
 void setup() {
   pinMode(LED_BUILTIN, OUTPUT);
 
@@ -74,6 +76,7 @@ void setup() {
   // so we can still log to the regular usbserial chips
   SoftwareSerial* ss = new SoftwareSerial(3, 1);
   ss->begin(SSBAUD);
+  ss->enableIntTx(false);
   logger = ss;
   logger->println();
   logger->printf("\n\nOn Software Serial for logging\n");
@@ -103,6 +106,8 @@ void setup() {
 
   timeout = (start_ms = last_ms = millis()) + TIMEOUT;
   logger->println("setup done");
+
+  deepSleepDeadline = ESP.getCycleCount() + ESP.getCpuFreqMHz() * 1000000 * 10;
 }
 
 void loop() {
@@ -196,4 +201,6 @@ void loop() {
         break;
       default:;
     }
+
+	if (static_cast<int32_t>(ESP.getCycleCount() - deepSleepDeadline) > 0) ESP.deepSleep(10000000);
 }

It works as expected, no WDT issues.
This is with esp8266 Arduino master HEAD and EspSoftwareSerial 5.0.0-rc3.

closed 2019-04-03 by dok-net