plerup/espsoftwareserial #83: Support ESP32 platform

iZhangHui, 2018-11-12:
ESP32 is more and more popular. Please add support for ESP32 platform. 
Currently, could't install this library with Arduino Library Manager.

Warlib1975, 2019-02-11:
Yes, please, register lib to be available via Arduino IDE and and ESP32 support. 

Thank you.

dok-net, 2019-03-23:
ESP32 working support, for what it's worth, is in release v4.0.0. Not sure about what to do regarding Arduino IDE built-in support, perhaps you could pursue that?

Warlib1975, 2019-03-23:
Now Arduino Library Manager doesn't show the library. Please, send them request to add the lib.

dok-net, 2019-03-24:
Can you a) perform tests to assure the next release is worthy, and give it a little time for everyone else to do the same? And b), could you please issue the request for inclusion of the latest release as an installable Arduino library at the appropriate time yourself? Please remember, this is all volunteer open and free software developed in our own spare time.

dok-net, 2019-04-03:
In cf2a93568ee485d1cff7a37082a1cb1478271638, compiles and passes basic Loopback to self test.

closed 2019-04-03 by dok-net