plerup/espsoftwareserial #85: Espsoftwareserial not thread safe

freeck, 2019-02-04:
During some stress tests I observed a lot of corrupted data on the receive side.
Test environment: an external serial device sends 1000 bytes messages at a baudrate of 115200, no pause between the characters.
After reviewing the library I discovered that the read and peek function are not thread safe. I expected an interrupt disable at the beginning and interrupt enable at the end of the fucntion, so the read pointer cant be modified during a the read or peak (these functions only take a few uS).
After that no corrupted data anymore.

dok-net, 2019-03-23:
This just might resolved by the new v4.0.0 code - could you please repeat your stress tests and report the results? Thanks!

freeck, 2019-03-26:
I tested this version today. Unfortunately no improvement. Even when using the basic example provided: "swsertest.ino" the messages are corrupted on random places within the message.
Can you explain why this version should be more efficiënt than the previous version?
I will try to increase the cpu-clock to 160MHz.....

dok-net, 2019-03-27:
An important caveat for everyone, the EspSoftwareSerial repository is in the esp8266com/esp8266 Git repository as a submodule. This is for the time being stuck at EspSoftwareSerial release 3.4.2!
At least my build environment ALWAYS picks up the submodule instead of any version in the workspace's libraries directory.
It's probably safe to delete the submodule checkout for the time being, as it's only used in test sketches of the esp8266 tree.

freeck, 2019-04-01:
Test environment: an external serial device sends 1000 bytes messages at a baud rate of 115200, no  pause between the characters. Message sent every 1 second.

I tested the following configurations:
Old version softwareSerial (thread safe version)
CPU at 80 MHz: error rate 25%
CPU at 160 MHz: error rate 3%

New version v4.0.0:
CPU at 80 MHz: error rate 70%
CPU at 160 MHz: error rate 20%

Looking at the code of the function int SoftwareSerial::available() and rxBits(); I am not surprised; I do not understand why this function has been made so complex! What is the philosophy of the code? 
Understandably I stick to the original version, see no reason to switch to v4.0.0 .

dok-net, 2019-04-01:
@freeck It makes a world of difference if you are using the latest releases of esp8266 for Arduino, 2.50 and onward including master's HEAD, and not the v4.0.0 of EspSoftwareSerial but maybe rather HEAD (or 5.0.0-rc1). Nevertheless, I'll try to find a way of reproducing your issue, or at least re-read the code with regard to it. Please hang in there.

freeck, 2019-04-01:
@dok-net I am using Sloeber Eclipse IDE to build my applications.  
The latest versions (2.5.0 beta 2 and 3) performed bad compared to esp8266-2.4.2, no idea Why but this was observed by others as well.
Which version are you using? Please send me the URL.

dok-net, 2019-04-02:
Modified loopback.ino:
---
#include <SoftwareSerial.h>

// local SoftwareSerial loopback, connect D5 (14) to D6 (12)

#ifdef ESP32
constexpr int SWSERBITRATE = 28800;
#else
constexpr int SWSERBITRATE = 115200;
#endif

constexpr SoftwareSerialConfig swSerialConfig = SWSERIAL_8N1;

constexpr int BLOCKSIZE = 1024; // use fractions of 256

SoftwareSerial swSerial(14, 12, false, 1040);

unsigned long start;
String effTxTxt("eff. tx: ");
String effRxTxt("eff. rx: ");
int txCount;
int rxCount;
int expected;
int rxErrors;
constexpr int ReportInterval = 10000;

Stream& logger(Serial);

void setup() {
	Serial.begin(115200);

	swSerial.begin(SWSERBITRATE, swSerialConfig);
	start = micros();
	txCount = 0;
	rxCount = 0;
	expected = -1;
	rxErrors = 0;
}

unsigned char c = 0;

void loop() {
	unsigned char block[BLOCKSIZE];
	for (int i = 0; i < BLOCKSIZE; ++i) {
		block[i] = c;
		c = (c + 1) % 256;
		++txCount;
	}
	swSerial.write(block, BLOCKSIZE);
	if (swSerial.overflow()) { logger.println("overflow"); }

	int avail;
	while (0 != (avail = swSerial.available())) {
		for (int i = 0; i < avail; ++i) {
			int r = swSerial.read();
			if (r == -1) { logger.println("read() == -1"); }
			if (expected == -1) { expected = r; }
			else {
				expected = (expected + 1) % 256;
			}
			if (r != (expected & ((1 << (5 + swSerialConfig % 4)) - 1))) {
				++rxErrors;
			}
			++rxCount;
		}
	}

	if (txCount >= ReportInterval) {
		logger.println(String("tx/rx: ") + txCount + "/" + rxCount);
		const auto end = micros();
		const unsigned long interval = end - start;
		const long txCps = txCount * (1000000.0 / interval);
		const long rxCps = rxCount * (1000000.0 / interval);
		const long errorCps = rxErrors * (1000000.0 / interval);
		logger.println(effTxTxt + 10 * txCps + "bps, "
					   + effRxTxt + 10 * rxCps + "bps, "
					   + errorCps + "cps errors (" + 100.0 * rxErrors / rxCount + "%)");
		start = end;
		txCount = 0;
		rxCount = 0;
		rxErrors = 0;
		expected = -1;
	}
	int rem;
	if (0 != (rem = swSerial.available())) {
		logger.print("swSerial.available() == ");
		logger.println(rem);
	}
}
---

EspSoftwareSerial latest release candidate:
https://github.com/plerup/espsoftwareserial/releases/tag/5.0.0-rc2

Esp8266/Arduino GitHub master:
https://github.com/esp8266/Arduino/commit/ca79f2ce39724527052319717a731b3501d5f0b2

Replace Esp8266/Arduino checkout in libraries/SoftwareSerial by the 5.0.0-rc2 release candidate.

While I think that 115200bps is stretching it a bit, actually I am getting very few errors in full-duplex loopback, running the ESP8266 at 160MHz. At 80MHz, 57600bps seems equal in error rate.

Could you please report your findings with hardware? I am also still performing various tests.

freeck, 2019-04-02:
Thanks for the info. I am still in the middle of a struggle getting the latest 2.5.0 at work. 
The example you use is not the same stress test I used; as it seems the data is sent synchronously and at a deterministic pace. I have simulated the external source using a Windows test programme, which generates buffered (1000 bytes) writes. You may then expect some timing differences which may explain an higher error rate. But I will continue tests.

I observe that using 160MHz clock is sufficient for my esp8266-app. So my issue is solved for now.
But what surprises me that latest versions (2.5.0 beta 2 and 3 and so on) performed worse compared to esp8266-2.4.2, no idea why but this was observed by others as well.

d-a-v, 2019-04-02:
2.5.0 is known to be (much) slower than 2.4.2 and git master version.
Use git if you can.
2.5.1 is expected soon (this month).

freeck, 2019-04-02:
Great! Thanks for the hint.

dok-net, 2019-04-03:
b34e7079d253e9133bce0830705e2f1ccf7f381c in my mind is the final release candidate. Changes since RC2 are basically cosmetic - except that the Loopback and Repeater tests have been revamped considerably for running two ESP8266 internal UARTs with each other. Sadly, they only manage 2400bps!
IIRC, esp8266 Arduino's release 2.4.x or 2.5.x was the initial reason for me to start work on EspSoftwareSerial, since my preferred application (SDS011 PM sensor, see https://github.com/dok-net/esp_sds011.git) didn't work at all once I updated. How well EspSoftwareSerial 4 and 5 performs on older esp8266 Arduino releases than 2.5.x, I don't know (and don't think it matters).
I may be mistaken, but you have to keep in mind that the ESP8266 is at heart a WLAN controller, and the cycles you get to use for Arduino sketches, and in the same way for interrupts etc., are a gift, not an entitlement. This goes to say that I believe 115200bps from any software serial emulation is nice to have,
but not a given, and subject to change with the releases from Espressif and the esp8266 Arduino project.
YMMV.

freeck, 2019-04-03:
I understand your points. For my case the older version of SoftwareSerial works nearly perfect at 115200 in receive-only mode, 1 channel.
 As the previous poster anounces the newer version 2.5.1 will be much Faster.
I did not dig deep into the 5.0.0-rc2 code but as earlier stated the code looks very inefficiënt to me.
That you only manage to transmit at 2400 baud in a loopback test doesnt surprise me. but I will compare the 2 versions with your example and report what happens.

dok-net, 2019-04-03:
To be sure, the 2400bps limit has nothing to do with EspSoftwareSerial, it's the performance in my tests between two ESP8266 modules crossover-connected between their internal hardware UARTS only! I have not cleaned up the two sketches very much, with all the preprocessor directives.
Returning to your concerns about the code in SoftwareSerial.cpp.
There are two requirements that I set out to fulfill. One, to make the write code as efficient as possible
with regards to sharing CPU cycles with other activity - that is the WLAN drivers and any other user tasks.
The other is to implement a best-practice interrupt service route, which does absolutely as little as
needed in the interrupt handler, and leaves everything else to userspace code. This is I have achieved.
For at least 9600bps you get quite usable full-duplex communication, which before was not possible as
far as my understanding of the blocking code in prior versions of EspSoftwareSerial goes.
Now, in order to free as much time between required real-time activity of the code, some analysis of the bit patterns takes place - this may look inefficient to you, but I am confident that it isn't. Possibly the waveform-like write timing generator may not be optimal, but then there's the option of requesting modifications from the waveform generation code in ESP8266 Arduino, like a period does not compulsively need to have both duty and off cycles, and does not terminate with low level on the GPIO, which currently stands in the way of generating the stop-bit level (high). I gave up on going that path for the time being.
So, while the releases from 4.0.0 onward may not provide much benefit for synchronous single-purpose single-task sketches, once you delve into async programming or multiple tasks, and full-duplex, you
should be starting to benefit from the new bits.
I hope the above provides some insight into the motivations and maybe you'll will find the new code
useful to your needs after all. Of course, you are free to stick to any previous version of the framework or
EspSoftwareSerial - I would be be quite interested in hearing if the pre-4.0.0 EspSoftwareSerial performs
adequately on current versions of the ESP8266 Arduino "OS".

freeck, 2019-04-03:
Ok, my assumption was obviously wrong concerning the test was wrong, sorry for that.
Thanks for your extensive explanation about your design Objectives.
I will keep you informed about my experiments with the Library.

Still one aspect stil not clear to me: do you recognize the read-function not being  "thread safe" ? Or are interrupts already blocked during the read-function?

dok-net, 2019-04-04:
Let's agree on something beforehand. The API is not explicitly re-entrant. Different SoftwareSerial instances must not get in another's way, of course.
Now, given the ESP8266 is single core, I don't recognize interrupt service routine modifications of data to be a multi-threading issue in the meaning of the word. I am quite surprised that disabling interrupts in read etcetera helped your data robustness, but I am not going to comment on the code before my
extensive modifications. That is because I have given the issues you are concerned about some thought,
especially as the ESP32 even is a multi-core CPU. Certainly not an expert on the IDF, I applied just
general knowledge to the design of interrupt service routines and preemptive multi-threaded multi-core
programming, and I used what C++ 11 (or later) as available in the current compiler offers.
All that said, the only relevant interactions where races could occur should be only within the rxRead() ISR, which can interrupt user space code at any time and which might get scheduled concurrently on
the second ESP32 core, and the rxBits() function, which are the only places that deal with the m_isrBuffer and supporting variables between them.
I believe that the memory fences I have used provide sufficient guarantees for the lock-free style to
prevent query corruption.
If you could point out anything that's wrong with that code, please by all means go ahead - disabling interrupts is not the correct answer, I am afraid, as that would be working against the whole idea of the separation of concern underlying the new receive code - it's bad enough already having to disable interrupts in order to improve writing timings to become sufficiently real-time at the temporary expense of full-duplex receiving.

freeck, 2019-04-04:
OK I will take your remarks in consideration. I try to reproduce the race condition and report on the results asap.

freeck, 2019-04-09:
FYI: I searched the internet on atomic-fences as you call them, related to interrupts.
I observe that esp8266 atomic objects may be interrupted. So in my case of continuously receiving and reading messages one might expect corruption of the pointer of the receive buffer.
See: https://github.com/wizard97/ArduinoRingBuffer/issues/2

And moreover: the IRAM usage of the rxRead() and available() functions is ca 1200 bytes! My app doesnt fitt anymore. 
Today I received #5965

dok-net, 2019-04-09:
I object to the notion of the (u)int32_t objects in use as being non-atomic on the ESP8266/ESP32. If you can, please provide working sample code or 3rd party discussion to the effect - my take on the ring buffer posts is the exact opposite of yours, namely, that the ESP8266 is fine and it's the 8bit Arduinos that may have issues.
I cannot figure out your statement on the rxRead() and available() functions. The former is tiny, and the latter is not attributed and thus should dynamically load from flash and be evicted from it on demand. Please, again provide working code samples to illustrate any issues.
Given that I am going forward with merging this submodule into Arduino esp8266, I am closing this issue. Please, by all means, if you are able to demonstrate failings, reopen a new issue with enough detail to allow use to work on solutions, and even better, send in some PR with fixes for material test cases.

closed 2019-04-09 by dok-net