plerup/espsoftwareserial #86: Unreliable loopback

larsch, 2019-03-23:
I'm trying v4.0.0, but I'm getting many errors, particularly on bytes with pattern 10000001, 11000001, 1110001, etc, i.e. when the last bits before the stop bit are set.

Sketch:

```c++
SoftwareSerial swSerial(RX_PIN, TX_PIN);

void setup()
{
  Serial.begin(115200);
  swSerial.begin(300);
}

void loop()
{
  static uint8_t val = 0;
  swSerial.write(val);
  auto timeout = millis() + 1000;
  while (!swSerial.available()) {
    if (millis() > timeout) {
      Serial.println("timeout\n");
      return;
    }
  }
  uint8_t rd = swSerial.read();
  if (rd != val) {
    Serial.printf("0x%02x != 0x%02x\r\n", rd, val);
  }
  ++val;
  if (val == 0x00)
    Serial.print('.');
}
```

Output (same result every run):

```
0x81: got 0xe0
0x82: got 0xf0
0x83: got 0xf0
0x84: got 0xf8
0x85: got 0xe1
0x86: got 0xf8
0x87: got 0xf8
0x88: got 0xfc
0x89: got 0xe2
0x8a: got 0xf1
0x8b: got 0xf1
0x8c: got 0xfc
0x8d: got 0xe3
0x8e: got 0xfc
0x8f: got 0xfc
0x90: got 0xfe
0x91: got 0xe4
0x92: got 0xf2
0x93: got 0xf2
0x94: got 0xf9
0x95: got 0xe5
0x96: got 0xf9
0x97: got 0xf9
0x98: got 0xfe
0x99: got 0xe6
0x9a: got 0xf3
0x9b: got 0xf3
0x9c: got 0xfe
0x9d: got 0xe7
0x9e: got 0xfe
0x9f: got 0xfe
0xa0: got 0xff
0xa1: got 0xe8
0xa2: got 0xf4
0xa3: got 0xf4
0xa4: got 0xfa
0xa5: got 0xe9
0xa6: got 0xfa
0xa7: got 0xfa
0xa8: got 0xfd
0xa9: got 0xea
0xaa: got 0xf5
0xab: got 0xf5
0xac: got 0xfd
0xad: got 0xeb
0xae: got 0xfd
0xaf: got 0xfd
0xb0: got 0xff
0xb1: got 0xec
0xb2: got 0xf6
0xb3: got 0xf6
0xb4: got 0xfb
0xb5: got 0xed
0xb6: got 0xfb
0xb7: got 0xfb
0xb8: got 0xff
0xb9: got 0xee
0xba: got 0xf7
0xbb: got 0xf7
0xbc: got 0xff
0xbd: got 0xef
0xbe: got 0xff
0xbf: got 0xff
0xc0: timeout
0xc1: got 0xf0
0xc2: got 0xf8
0xc3: got 0xf8
0xc4: got 0xfc
0xc5: got 0xf1
0xc6: got 0xfc
0xc7: got 0xfc
0xc8: got 0xfe
0xc9: got 0xf2
0xca: got 0xf9
0xcb: got 0xf9
0xcc: got 0xfe
0xcd: got 0xf3
0xce: got 0xfe
0xcf: got 0xfe
0xd0: got 0xff
0xd1: got 0xf4
0xd2: got 0xfa
0xd3: got 0xfa
0xd4: got 0xfd
0xd5: got 0xf5
0xd6: got 0xfd
0xd7: got 0xfd
0xd8: got 0xff
0xd9: got 0xf6
0xda: got 0xfb
0xdb: got 0xfb
0xdc: got 0xff
0xdd: got 0xf7
0xde: got 0xff
0xdf: got 0xff
0xe0: timeout
0xe1: got 0xf8
0xe2: got 0xfc
0xe3: got 0xfc
0xe4: got 0xfe
0xe5: got 0xf9
0xe6: got 0xfe
0xe7: got 0xfe
0xe8: got 0xff
0xe9: got 0xfa
0xea: got 0xfd
0xeb: got 0xfd
0xec: got 0xff
0xed: got 0xfb
0xee: got 0xff
0xef: got 0xff
0xf0: timeout
0xf1: got 0xfc
0xf2: got 0xfe
0xf3: got 0xfe
0xf4: got 0xff
0xf5: got 0xfd
0xf6: got 0xff
0xf7: got 0xff
0xf8: timeout
0xf9: got 0xfe
0xfa: got 0xff
0xfb: got 0xff
0xfc: timeout
0xfd: got 0xff
0xfe: timeout
0xff: timeout
```

dok-net, 2019-03-24:
300bps was never anything I'd have considered testing, in the first place.
But nevermind, you have to keep in mind that the ESP8266 is first and foremost busy keeping WLAN
running, besides WDT panics there are other ways of causing havoc when not delay-ing frequently and long enough. This is the cause of the problem you are reporting, please insert a sufficiently long delay(...) in the available() loop.
Tested and found working without flaws, closing issue.

closed 2019-03-24 by dok-net

larsch, 2019-03-24:
Many utility meters (electricity, heating) use 300 bps communication via IR.

First, I don't have WiFi running and there are no WDT resets.

Second, the bits should already be available while Tx'ing (its a loopback). The algorithm should be able to pick them up. There should be no reason to delay. I think the algorithm is buggy; my own implementation has no problems receiving at 300 bps in a busy loop (keeping the WDT fed when doing many bytes).

dok-net, 2019-03-26:
The fixed code works without any extra yields, delays etc. in the sketch on the ESP8266 as well.