plerup/espsoftwareserial #87: ESP8266 - void sws_isr() [with int I = 9] causes a section type conflict

IoTNotes, 2019-03-31:
When I try to compile on Esp8266 I get these errors:

Compiling 'Fiverr_proj_01' for 'NodeMCU 1.0 (ESP-12E Module)'
 
SoftwareSerial.cpp: In function void sws_isr() [with int I = 9]
 
SoftwareSerial.cpp: 40:38: error: void sws_isr() [with int I = 9] causes a section type conflict with virtual size_t SoftwareSerial::write(uint8_t)
   template<int I> void ICACHE_RAM_ATTR sws_isr() {
SoftwareSerial.cpp:284: note  virtual size_t SoftwareSerial  write(uint8_t) was declared here
 size_t ICACHE_RAM_ATTR SoftwareSerial*: write(uint8_t b) {
Error compiling libraries
Build failed for project 'Fiverr_proj_01'

dok-net, 2019-04-01:
I cannot reproduce this issue.
I heavily suspect that it is caused by the blending of two different versions of EspSoftwareSerial, one, that you use on purpose and which is the latest, and the other being introduced through the back door by the esp8266com/esp8266 repository, which uses EspSoftwareSerial as a submodule in libraries/SoftwareSerial and that is currently stuck at release tag 3.4.2.

IoTNotes, 2019-04-01:
Well, I suspected a conflict between SoftwareSerial esp8266 default library and espSoftwareSerial Library. This is why I deleted the original library and substituted with espSoftwareLibrary renamed to SoftwareSerial.
But I still have this error.
How do you suggest to install espSoftwareSerial?

d-a-v, 2019-04-01:
@IoTNotes 

To update esp8266/arduino SoftwareSerial submodule to lastest master:
```
(optional, clean it:)
$ rm -rf libraries/SoftwareSerial
$ git submodule update --init
(now update it:)
$ cd libraries/SoftwareSerial
$ git checkout master
$ git pull
```

@dok-net 

> [...] and the other being introduced **through the back door** by the esp8266com/esp8266 repository
> [...] currently **stuck** at release tag 3.4.2.

I don't understand this way of writing.
I was the one who introduced EspSoftwareSerial as a submodule in the main esp8266/arduino repository. Is that a betrayal ? I was thinking the contrary at that time, when I pushed the latest stable version which was also the git head version.

Now you are a maintainer and actively working on it. And I find it great you do this.
Latest improved stable version is about to change thanks to you - and I know how much time you tried to push your work. I even recently sent you a congrat email about that. 

You can always reproduce the steps above on your esp8266/arduino fork, and propose a pull request on the main repository.
Would it be fair for esp8266/arduino users to follow every single commit of EspSoftwareSerial, or is it more legal to update once a stable version is published ?

Suggestion: If you agree with the steps above, a section in your main README.md mentionning them would be useful for esp8266/arduino git users so they can update themselves at your latest commits if they want to.

Why did I add it as a submodule:
- because EspSoftwareSerial was shipped with packaged versions of esp8266/arduino
  but *not* with the git version
- because we use it in some examples (one example I added)
- because I found it had its place there

dok-net, 2019-04-01:
Dear d-a-v,
it should for obvious reasons not be my intention to alienate collaborators, and it surely was not, please accept my sincerest apologies, I can see now that the wording could be seen as destructive criticism, but wasn't meant to be, instead just for me the shortest possible way of stating that I also got caught by surprise that changes to EspSoftwareSerial never materialized in build runs until I disovered it in the esp8266 repo as a submodule.
That said, let me repeat my reservations about the use of submodules in general, which are founded in the ambiguities expressed right in the git documentation. But, I really don't know better, and it's not unusual.
I have now amended the README.md as you suggested.

d-a-v, 2019-04-01:
> [...] until I disovered it in the esp8266 repo as a submodule.

Yes: EspSoftwareSerial is shipped in the esp8266/arduino releases since some years.
Given the very easy way to update a submodule (that you just added in the documentation), I found it fair to put release users and git users on the same line.

I was aware that it would create some conflicts at the beginning, and sorry about that. I can say now there was not too much complaints about it (given the submodule was pointing to the latest version).

> [...] which are founded in the ambiguities expressed right in the git documentation.

I'm not aware about that. My thinking is that submodules is a very nice way to track a specific version of a repository, specially in its stable state. It can be updated locally to any version, like HEAD, and easily updatable with a simple commit of the updated subdirectory in the main repository.

(@IoTNotes sorry to pollute your issue :)

dok-net, 2019-04-01:
@IoTNotes with all the new information, are you still getting the same result? Possibly a stupid question, but are you working with the master HEAD of the esp8266 repo and have you remembered to run tools/get.py also?

IoTNotes, 2019-04-01:
@d-a-v, thankyou for your support.
I apologize, be patient, but I'm not a github lover, so I'm used  to clone a repository downloading the zip file and then extracting the file into the target directory.
I really don't understand a lot about submodules and cli git commands....... I know, it's not usual in these days and this is why I have always in my to-do list "Github learning".
I downloaded last repository from master, deleted everything inside SoftwareSerial directory and extracted all the zipped files in it. In this way I suppose to work with the latest version of espSoftwareSerial but maybe I'm wrong, because I get always the same error compiling examples/swsertest.ino with a Nodemcu v1.0 (ESP-12E module).
Thanks in advance.

d-a-v, 2019-04-01:
> be patient

That I am everyday, like any maintainer needs to be :)

Once you've familiarized with git and submodules (not specifically with github), what you did by hand (the right thing) will be a matter of one line (or one click if a gui is in the way).

I suspect you are not using a recent version of the esp8266 arduino core.

IoTNotes, 2019-04-01:
Damn... as easy as stupid (me!)
I was using esp8266 core 2.4.x, updated to 2.5.0 and compiles straight!
Thanks a lot for your super fast support and your "typical mainteiner" patience! :-)

closed 2019-04-01 by dok-net

jLynx, 2019-04-04:
Is there a version that supports 2.4.x? As I am limited to this version for my current project

dok-net, 2019-04-04:
You can always use the EspSoftwareSerial version that was current at esp8266 Arduino 2.4.2. That's 3.4.3, https://github.com/plerup/espsoftwareserial/commit/56654d24d09c177a5110418d4cc6acf87bdece2b. Or you can also go with 4.0.0, https://github.com/plerup/espsoftwareserial/commit/fc38a0f6dd82be24f0bd186b8e7e26b62ad2383c.
I don't know about the support policy for esp8266 Arduino, and a few issues with the new features in
EspSoftwareSerial were also fixed between those releases and the upcoming 5.0.0.
What's behind your having to stick with 2.4.x?

puthre, 2019-04-15:
I'm getting the same error using esp8266 core 2.5.0 and espsoftwareserial 5.0.3 from git
Tried with different core versions and the espsoftwareserial that can be installed from arduino IDE, same thing.

/home/puthre/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/puthre/arduino-1.8.5/hardware -hardware /root/.arduino15/packages -hardware /root/Arduino/hardware -tools /home/puthre/arduino-1.8.5/tools-builder -tools /home/puthre/arduino-1.8.5/hardware/tools/avr -tools /root/.arduino15/packages -built-in-libraries /home/puthre/arduino-1.8.5/libraries -libraries /root/Arduino/libraries -fqbn=esp8266com:esp8266:nodemcu:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path /tmp/arduino_build_612972 -warnings=none -build-cache /tmp/arduino_cache_795453 -prefs=build.warn_data_percentage=75 -verbose /root/Arduino/softwareseria/sketch_apr15a/sketch_apr15a.ino
/home/puthre/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/puthre/arduino-1.8.5/hardware -hardware /root/.arduino15/packages -hardware /root/Arduino/hardware -tools /home/puthre/arduino-1.8.5/tools-builder -tools /home/puthre/arduino-1.8.5/hardware/tools/avr -tools /root/.arduino15/packages -built-in-libraries /home/puthre/arduino-1.8.5/libraries -libraries /root/Arduino/libraries -fqbn=esp8266com:esp8266:nodemcu:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path /tmp/arduino_build_612972 -warnings=none -build-cache /tmp/arduino_cache_795453 -prefs=build.warn_data_percentage=75 -verbose /root/Arduino/softwareseria/sketch_apr15a/sketch_apr15a.ino
Using board 'nodemcu' from platform in folder: /home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266
Using core 'esp8266' from platform in folder: /home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266
Build options changed, rebuilding all
Detecting libraries used...
"/root/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/lwip2/include" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_612972/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/variants/nodemcu" "/tmp/arduino_build_612972/sketch/sketch_apr15a.ino.cpp" -o "/dev/null"
"/root/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/lwip2/include" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_612972/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/variants/nodemcu" "-I/root/Arduino/libraries/espsoftwareserial/src" "/tmp/arduino_build_612972/sketch/sketch_apr15a.ino.cpp" -o "/dev/null"
"/root/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/lwip2/include" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_612972/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/variants/nodemcu" "-I/root/Arduino/libraries/espsoftwareserial/src" "/root/Arduino/libraries/espsoftwareserial/src/SoftwareSerial.cpp" -o "/dev/null"
Generating function prototypes...
"/root/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/lwip2/include" "-I/home/puthre/arduino-1.8.5/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" [... cut ...]
/root/Arduino/libraries/espsoftwareserial/src/SoftwareSerial.cpp:36:38: error: void sws_isr() [with int I = 9] causes a section type conflict with virtual size_t SoftwareSerial::write(uint8_t)
 template<int I> void ICACHE_RAM_ATTR sws_isr() {
                                      ^
/root/Arduino/libraries/espsoftwareserial/src/SoftwareSerial.cpp:256:24: note: 'virtual size_t SoftwareSerial::write(uint8_t)' was declared here
 size_t ICACHE_RAM_ATTR SoftwareSerial::write(uint8_t b) {
                        ^
Using library espsoftwareserial at version 5.0.3 in folder: /root/Arduino/libraries/espsoftwareserial 
exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).