plerup/espsoftwareserial #89: esp8266/arduino example @115200bauds: first character of a printf is incorrect

d-a-v, 2019-04-01:
When running current master of this library with esp8266/arduino's `SerialStress` core example, where EspSoftwareSerial is used for logging @115200 bauds, first character of the log line is always incorrect.

I understand this baudrate is near to the acceptable limit, however,
- this is always happen on the first character, not (always) on others, so I naively tend to think it could be fixed
- this behaviour is not reproducible with 3.4.3 stable version shipped with esp8266/arduino
  (even though with 3.4.3 some other characters can be missed too, like below)

```
10:37:40.762 -> `wavg=2392 bwnow=2391 kbps mapavail=217
10:37:41.724 -> `wavg=2392 bwnow=2394 kbps maxavail=217
10:37:42.718 -> `wavg=2392 bwnow=2391 kbps maxavail=217
10:37:43.680 -> `wavg=2392 bwnow=2391 kbps mapavail=217
10:37:44.674 -> `wavg=2392 bwnow=2394 kbps maxavail=217
10:37:45.669 -> `wavg=2392 bwnow=2391 kbps mapavail=217
10:37:46.631 -> `wavg=2392 bwnow=2394 kbps maxavail=217
10:37:47.592 -> `wavg=2392 bwnow=2391 kbps mapavail=217
10:37:48.587 -> `wavg=2392 bwnow=2391 kb`s maxavail9217
10:37:49.552 -> `wavg=2392 bwnow=2394 kbps maxavail=217
10:37:50.546 -> `wavg=2392 bwnow=2391 kbps maxavail=217
10:37:51.508 -> `wavg=2392 bwnow=2394 kbps maxavail=217
```

(should be `bwavg=`, `b` is 98=`0b01100010` and ` `` ` is 96=`0b01100000`)

dok-net, 2019-04-01:
I am completely unable to reproduce your reported issue :-(

In 4 Wemos D1 mini, 3 of them with different hardware:

USB-SERIAL CH340:
Uploading to I/O board
Opening port
Port open
bwavg=2400 bwnow=2400 kbps maxavail=160
bwavg=2397 bwnow=2394 kbps maxavail=160
bwavg=2396 bwnow=2394 kbps maxavail=161
bwavg=2396 bwnow=2396 kbps maxavail=161
bwavg=2396 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2396 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2396 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
[...]

USB-SERIAL CH340:
Opening port
Port open
bwavg=2398 bwnow=2398 kbps maxavail=143
bwavg=2397 bwnow=2396 kbps maxavail=160
bwavg=2396 bwnow=2394 kbps maxavail=161
bwavg=2396 bwnow=2396 kbps maxavail=161
bwavg=2396 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2396 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2396 kbps maxavail=161
bwavg=2385 bwnow=2394 kbps maxavail=161
[...]

USB-SERIAL CH340:
Opening port
Port open
bwavg=2400 bwnow=2400 kbps maxavail=142
bwavg=2397 bwnow=2394 kbps maxavail=160
bwavg=2396 bwnow=2394 kbps maxavail=161
bwavg=2396 bwnow=2396 kbps maxavail=161
bwavg=2396 bwnow=2394 kbps maxavail=161
bwavg=2396 bwnow=2396 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2396 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2396 kbps maxavail=161
[...]

Silicon Labs CP210x USB to UART Bridge:
Opening port
Port open
bwavg=2400 bwnow=2400 kbps maxavail=160
bwavg=2397 bwnow=2394 kbps maxavail=160
bwavg=2397 bwnow=2396 kbps maxavail=161
bwavg=2396 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2396 bwnow=2396 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2396 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2396 kbps maxavail=161
bwavg=2395 bwnow=2394 kbps maxavail=161
bwavg=2395 bwnow=2396 kbps maxavail=161
[...]

dok-net, 2019-04-01:
Things look dramatically different at 160MHz SOC clock:

Silicon Labs CP210x USB to UART Bridge:
Opening port
Port open
b�@�V����ݹ���2402 kbps maxavail=100
�Tߝ�2398 bwnow=2394 kbps maxavail=151
�watg=2397 bwnow=2396 kbps maxavail=152
�Tߝ�2397 bwnow=2396 kbps maxavail=152
�Tߝ�2397 bwnow=2396 kbps maxavail=152
�Wѝ�2396 bwnow=2394 kbps maxavail=152
�Tߝ�2396 bwnow=2396 kbps maxavail=152
`Wѝ�2396 bwnow=2396 kbps maxavail=152
�Tߝ�2396 bwnow=2396 kbps maxavail=152
�T���2396 bwnow=2396 kbps maxavail=152
`Wѝ�2396 bwnow=2394 kbps maxavail=152
[...]

USB-SERIAL CH340:
Opening port
Port open
b�@vg=2400 bwnow=2400 kbps maxavail=100
�Tߝ�2398 bwnow=2396 kbps maxavail=151
bWѝ�2397 bwnow=2396 kbps maxavail=152
�wapg=2397 bwnow=2396 kbps maxavail=152
�Tߝ�2396 bwnow=2394 kbps maxavail=152
bWѝ�2396 bwnow=2396 kbps maxavail=152
�Gapg=2396 bwnow=2396 kbps maxavail=152
`savg=2396 bwnow=2396 kbps maxavail=152
��avg=2396 bwnow=2396 kbps maxavail=152
�wapg=2396 bwnow=2394 kbps maxavail=152
`Wٝ�2396 bwnow=2396 kbps maxavail=152
�Gapg=2396 bwnow=2396 kbps maxavail=152

d-a-v, 2019-04-01:
For the result above, I am using 80Mhz, an esp-01 with an ftdi converter and linux host. I had the same results as yours at 160Mhz.
So I also tried with a D1 mini + CH340 on the same host and I have better results (average 50% of output is good).
I'm not saying it should work 100% and probably the example should use a lower softserial baudrate.

I was (and now am) wondering why
- if only the first char has an issue, is there something that can be done about it (I'm aware it will be difficult for you if you can't reproduce :)
- why 80MHz gives better result than 160MHz

dok-net, 2019-04-01:
Please add in SerialStress.ino, after ss->begin(...):
ss->enableIntTx(false);

And report your findings at 80MHz.
As I am not seeing bad bits in the first few octets at 80MHz, but at 160MHz, and the above change alone
doesn't solve the issue for me at 160MHz (but not reenabling IRs in SoftwareSerial::preciseDelay does), I
am really surprised that the higher clock rate causes more harm if some interrupts are allowed during
the delays for the bit timing at 160MHz but not at 80MHz.......

d-a-v, 2019-04-01:
With 
```
  SoftwareSerial* ss = new SoftwareSerial(3, 1);
  ss->begin(SSBAUD);
  ss->enableIntTx(false);
```
There is no notable change with both 80 and 160 MHz

d-a-v, 2019-04-02:
Thanks, from my POV,
Latest commit is good at 80MHz
It is also good @160Mhz *and* with `ss->enableIntTx(false);`

dok-net, 2019-04-02:
@d-a-v 
Please try b34e7079d253e9133bce0830705e2f1ccf7f381c
Same behavior as you commented on, but not due to arbitrary "51" µs hack.

d-a-v, 2019-04-02:
@dok-net 
I tried https://github.com/plerup/espsoftwareserial/commit/b34e7079d253e9133bce0830705e2f1ccf7f381c (from your own repo) and it is worse than current master, at 80 or 160MHz, with or without `ss->enableIntTx(false);` (not a single line is right)

dok-net, 2019-04-02:
@d-a-v wtf?
I've just this minute rebuilt and flashed to an arbitrary Wemos D1 mini, 80MHz, SerialStress.ino with the

ss->begin(SSBAUD);
ss->enableIntTx(false);
logger = ss;

modification (nothing else is changed), and I am getting in my terminal window, when I press the reset button on the ESP8266:

On Software Serial for logging
SDK:2.2.1(cfd48f3)/Core:unix-2.6.0-dev=-194000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-2-ga501b57/BearSSL:6778687

BAUD: 3000000 - CoreRxBuffer: 1024 bytes - TestBuffer: 4096 bytes
led changes state every 300000 bytes (= 1 second)
press 's' to stop reading, not writing (induces overrun)
press 't' to toggle timeout testing on readBqtes
setup done
bwavg=2401 bwnow=2401 kbps maxavail=160
bwavg=2397 bwnow=2394 kbps maxavail=163
bwavg=2397 bwnow=2396 kbps maxavail=163
bwavg=2396 bwnow=2394 kbps maxavail=163
bwavg=2396 bwnow=2394 kbps maxavail=190
bwavg=2395 bwnow=2394 kbps maxavail=190
bwavg=2395 bwnow=2396 kbps maxavail=190
bwavg=2395 bwnow=2394 kbps maxavail=190
bwavg=2395 bwnow=2394 kbps maxavail=206
bwavg=2395 bwnow=2396 kbps maxavail=206
bwavg=2395 bwnow=2394 kbps maxavail=206
bwavg=2395 bwnow=2394 kbps maxavail=206
bwavg=2395 bwnow=2394 kbps maxavail=206
bwavg=2395 bwnow=2396 kbps maxavail=206
bwavg=2395 bwnow=2394 kbps maxavail=206
bwavg=2395 bwnow=2394 kbps maxavail=206
bwavg=2395 bwnow=2396 kbps maxavail=206
bwavg=2395 bwnow=2394 kbps maxavail=206

d-a-v, 2019-04-02:
With your own repo's master @80MHz original sketch 
latest commit:
```
commit b34e7079d253e9133bce0830705e2f1ccf7f381c (HEAD, doknet/master, master)
Author: Dirk O. Kaar <dok@dok-net.net>
Date:   Tue Apr 2 16:01:36 2019 +0200

    Relaxed timing requires reset of deadline, with or without interrupts.
```
I get
```
17:32:28.686 -> ⸮wavg=2393 bwnow=2391 kbps maxavail9197
17:32:29.681 -> ⸮wavg=2313 bwnow=2311 kbps maxavail=117
17:32:30.642 -> ⸮wavg=2313 bwnnw=2314 kbps maxavail=117
17:32:31.636 -> ⸮wavg=2313 bwnow=2311 kbps maxavail=197
17:32:32.617 -> ⸮wavg=2313 bwnow=2311 kbps maxavail=117
17:32:33.590 -> ⸮wavg=2313 bwnnw=2314 kbps maxavail=197
17:32:34.552 -> ⸮wavg=2313 bwnow=2311 kbps maxavail=117
17:32:35.546 -> ⸮wavg=2312 bwnnw=2312 kbps maxavail=117
```

Back to master of this repo,
latest commit:
```
commit 34893752d8dad9dc93a891d146186ada60a473a9 (HEAD -> master, origin/master, origin/HEAD)
Author: Dirk O. Kaar <dok@dok-net.net>
Date:   Mon Apr 1 15:13:52 2019 +0200

    Empirically determined a minimum micro second delay that can occur due to reenabled interrupts.
    
    For clean transmit timing, only reenable interrupts if delay exceeds that duration.
    Works at 80MHz and 160MHz - only tested with the ESP8266 SerialStress sketch at this moment.
    Pending further tests.
    
    Affects (#89).

```
I get:
```
17:36:02.287 -> bwavg=2398 bwnow=2398 kbps maxavail=161
17:36:03.280 -> bwavg=2395 bwnow=2391 kbps maxavail=161
17:36:04.241 -> bwavg=2394 bwnow=2394 kbps maxavail=171
17:36:05.235 -> bwavg=2394 bwnow=2391 kbps maxavail=171
17:36:06.228 -> bwavg=2393 bwnow=2391 kbps maxavail=171
```

dok-net, 2019-04-03:
b34e7079d253e9133bce0830705e2f1ccf7f381c is probably the final release candidate as far as I am concerned. I'm maxxed out and given the same applies even to the ESP2866's internal UART's capacity to interconnect to another ESP8266 with the same 20cm leads I'm using for testing EspSoftwareSerial, AND that we are honestly seeing different results on what the above test yields in what I consider a corner case at least for anything but simplex communication (115200bps for SW serial on the ESP8266 of all WLAN (!) microcontrollers), this is as good as it's going to be.
Naturally, I will be happy to consider any PRs with improvents.

d-a-v, 2019-04-03:
Current master (cf2a93568ee485d1cff7a37082a1cb1478271638) is running nicely.
I retried with b34e7079d253e9133bce0830705e2f1ccf7f381c and it seems running nicely too with the exact same hardware as earlier in this thread. I have no explanation (I had triple-checked my results). 
I understand we are playing with limits with this baudrate.
Thanks, closing, I encourage you to make a PR on esp8266 arduino core repository with your latest release

closed 2019-04-03 by d-a-v

oizodzil357, 2019-04-18:
Why not choosing 9600 bauds rate ?

d-a-v, 2019-04-18:
Because all examples in esp8266/arduino are @115200, and the example involving software serial uses the standard hardware serial pins.
The example is supposed to show how to replace hardware serial logging sketches by a software serial one while using hardwareserial at higher baudrates.
It is okay to get bad characters sometimes at this baudrate, it is for logging in this particular example. As an example it is also good to show users the limitations of softwareserial.
Getting to limits is also good to find issues, it also improve lower rates robustness. Thanks @dok-net :)