plerup/espsoftwareserial #92: Arduino IDE using both ESPSoftwareSerial and SoftwareSerial

resch59, 2019-04-18:
I am attempting to use the ESP8266 with the IDE version 1.8.7.

 I am working on several projects at the moment and one is with the ESP8266 using ESPSoftwareSerial. I am also working on a project with an Mini Pro using the standard SoftwareSerial.

Because both programs use "#include SoftwareSerial.h" I have to remove the ESP version from the library folder for the Mini Pro project to compile with no errors. When I work on the ESP project I have to remove the SoftwareSerial folder from the library and move the ESP folder back.

Because i am not completely familiar with using the Arduini IDE I would like not to have to keep swapping the libraies. Can you give me some advice on how I can keep both folder in the library folder?

dok-net, 2019-04-19:
With current versions of the respective libraries, the Arduino environment should pick out the correct libraries based on the supported hardware. I cannot confirm your report - are you speaking from expectation of errors or are you really seeing conflicting #includes and compilation units? I am inclining toward rejecting this issue.

resch59, 2019-04-19:
Sorry but I am some new to the arduino. Yes I get compiler errors. It always looks for the standard SoftwareSerial library when I try and compile esp program. If I remove it from library folder it uses the espsoftwareserial library. The errors indicate there are multiple softwareserial.h files and show both but tries to use the standard version. I am away for a few days to see family over Easter. When I return I can send screen shot of errors.

dok-net, 2019-04-19:
Arduino IDE 1.8.9 from https://www.arduino.cc/en/Main/Software. Everything updated through board and library manager. SoftwareSerial for AVR is integral part of the AVR boards package, the EspSoftwareSerial library is installed via library manager, the ESP8266 Arduino is HEAD from GitHub. 

Simple sketch:
---
#include <SoftwareSerial.h>
SoftwareSerial swSer(12, 14);
void setup() {
  Serial.begin(9600);
  swSer.begin(9600);
}
void loop() {
  int swIn = swSer.read();
  if (swIn >= 0) Serial.print(swIn);
  int hwIn = Serial.read();
  if (hwIn >= 0) swSer.print(hwIn);
  // swSer.perform_work();
}
---
Through menu Tools/Board I can switch between "Arduino Pro or Pro Mini" and "Generic ESP8266 Module" at will, "Verify" compiles without issues for either board. Also, I have used the swSer.perform_work() line to prove that for each board, either the Arduino AVR SoftwareSerial.h or the EspSoftwareSerial SoftwareSerial.h gets used.

I am afraid, I am not able to provide assistance beyond this, should there be an issue with the Arduino IDE version you are relying on, or perhaps the release version of the ESP8266 Arduino, if you could narrow down your issue, maybe someone from those communities can give some advice?

resch59, 2019-04-22:
back from easter break. I found my problem. I reinstalled the IDE and now it is working. Thanks for help and suggestions.

closed 2019-04-22 by resch59