plerup/espsoftwareserial #93: SWSERIAL_9N1 Possible?

ondabeach, 2019-04-20:
Hi, thanks for this beautiful code. It is the only softwareserial that will work on the ESP32. But I need 9 data bits, is it possible to add this config mode?. I know it means doubling the buffer size but that does not matter for me. I looked at the code in espsoftwareserial and it is beyond my level of understanding, otherwise I would try to do it myself.

Thanks, Steve.

dok-net, 2019-04-23:
@Steve, discussion regarding your requirement happens in https://github.com/espressif/arduino-esp32/pull/2701

dok-net, 2019-04-23:
@ondabeach, to the point of code cleanups I have done some preliminary work to allow one to identify the spots that need to be touched for multi-byte words. In my spare time, I am afraid, I am not going to do much more than that. As were are learning, the ESP32 team don't have much love for software serial emulation, either. If you want/need this mode commercially, a different arrangement should be possible.

fredvik, 2019-05-11:
I've adapted an older version of this library to support parity (None, Odd, Even, Mark and Space). I'm using it to detect and send addressing commands using mark/space parity in a RS485 protocol. It should be sufficient to solve most "9-bit serial problems" I've seen people asking about.
 
@ondabeach - What is your use case, do you need 9 true data bits or is the 9th bit used separately, e.g. for addressing in the protocol?

Unfortunately my application isn't very stable and I don't know if the problem is that other parts of my code interferes with realtime requirements (most likely), my patches to ESPSoftwareSerial or the library itself (unlikely).

If flexible parity options can be added it could be a great selling point for this library. As far as I can tell there is no simple way to implement Space and Mark parity in hardware for ESP8266 or ESP32 (but I'd love to be corrected).

It's great to see that so much work has been done here over the last year and I'm hoping that the updated code will be more forgiving to my application and can improve overall stability for me.

So I am currently on track to adapt my parity code to the curent ESPSoftwareSerial layout and plan to share it with you all. It will be a slow process with no guarantees as it is for a small hobby project. Also, I am not a professional programmer by any means, so keep your expectations low :-)

Obviously, I would love to know if anyone else is also working on it.

dok-net, 2019-05-11:
@fredvik using explicit mark/space parity, as I assume combined with "manually" shifting the data bits in user code to simulate 9bits on the wire, is something that I've been thinking about, too. If there is some standard API to this, accepting PRs to this end would be easier.
That said, I am still dealing with some problems seemingly related to the extremely sensitive GPIOs on ESP32 - it's basically able to detect signals over an air-gap and it suffers significantly in full-duplex if adjacent pins are used - but even pins on opposite edges of the breakout boards don't suffice for proper bit detection. I am currently investigating debouncing, frequency analysis, etc. - in short, the code is in constant flow, not a very good time to accept PRs. You are always welcome to do your own patches and share them, naturally, just be prepared that merging them anytime may be some effort again.
Please check out my working repository https://github.com/dok-net/espsoftwareserial

fredvik, 2019-05-12:
@dok-net That's fine with me! I'll have a look at your working repository as well.
In general it should be pretty straightforward to support parity for Tx. It can easily be added on the fly when start+data+parity+stop bits are pre-loaded, just before bit banging begins. Full support for Rx will likely require more substantial changes due to buffering. Perhaps that's also where reusing a common api, e.g. for managing parity errors, is most useful.
I've only seen one or two libraries for arduino with (partial) parity support and I'll definitely do some more research to see if there is some default api that can be reused.

dok-net, 2019-11-12:
@fredvik Keeping the API you designed in spirit, I've merged your README enhancements and implemented the "9bit" as parity with mark and space extension. It would be great if could have a moment or two available to try and let me know if it works for you, too.
Naturally, I've tested with the internal HW UART, on both ESP8266 and ESP32.

closed 2019-11-12 by dok-net