plerup/espsoftwareserial #94: Parity...

kg9316, 2019-04-27:
Thanks for a great library !
Is it possible to implement EVEN parity ?

dok-net, 2019-04-27:
Possible, yes, naturally. Reasonable, I am not sure. Someone has to spend time on doing this... Have you considered Serial.swap() to redirect the HW serial to D7 (Rx, GPIO13) / D8 (Tx, GPIO15) on the ESP8266 and use EspSoftwareSerial for logging? The ESP32 has three full UARTs, for that matter.

kbickham, 2019-07-30:
I think this is unrealistic for the timescale I have, but the software I2c on esp8266 is ..well, not. I'm trying to use AT commands to an AS7263 6 channel spectrometer, but the protocol is start, 8, parity, stop.Can't swap because I'm using hw serial for comms with an avr. Good work though man, respect.

dok-net, 2019-07-30:
@kbickham Can't you use SoftwareSerial with the AVR and free the HardwareSerial to `swap()`?

kbickham, 2019-07-30:
You're going to laugh but I'm already using https://github.com/madsci1016/Arduino-EasyTransfer to pass over the state of a machine (like 100 bytes). Copying over pull 101 to see if it'll talk, if not... I'll try. I do believe that protocol is simpler, or at the least I could mess with it to make it manageable. That's a good idea.

kbickham, 2019-07-30:
Yep. I've wasted enough time. Good call.

dmascord, 2019-07-30:
Hi,

I have used i2c on micropython on esp8266 with great success, would that work for your use case?

Cheers,

Damien

dok-net, 2019-07-30:
Project pitch: [A co-operative stackful portable multitasking library for ESP8266, ESP32, Arduino boards, Windows, Linux](https://github.com/dok-net/CoopTask/releases/tag/1.3.0)
Put separate concerns of your MCU programs into different tasks, synchronize and exchange data where necessary, re-use code more easily.

kbickham, 2019-07-30:
I'm using firmware I wrote in c++ on it, to complement my main board. It's
basically a state machine that reads from a bunch of stuff so I'm tied to
that I believe. i have an esp 32 here... i'm an idiot i just need to use
that.

dok-net, 2019-11-12:
Implemented in release 6.1.0. Please test and report.

closed 2019-11-12 by dok-net

kbickham, 2019-11-12:
Yes. Actually implemented this on my senior design project. It is in fact
stable, while also using old dht library ( wierd 70 and 20ns timing) in a
non blocking ladder/switch case. Sorry I didn't report sooner.

dok-net, 2019-11-12:
I think there's a misunderstanding. I've now a cleanroom implementation in the official EspSoftwareSerial. It would be great if you had any hardware to test against.Sent from my Huawei Mobile

kbickham, 2019-11-12:
My apologies, I'm actually using a different library. I don't want to blow up a closed thread, but if you look at the esp8266 code on my github, that's what I'm using in this https://www.youtube.com/watch?v=yx6x5Qz6MVk. It was stable enough to implement EasyTransfer library to send structs between an 8266 and esp328.