plerup/espsoftwareserial #95: attachInterruptArg is not available for ESP8266

saeugetier, 2019-04-27:
EspSoftwareSerial doesn't compile with Arduino ESP8266 anymore. The function "attachInterruptArg" doesn't exist in the ESP8266 Arduino Core.

See comment: https://github.com/plerup/espsoftwareserial/commit/61d18be74a0511c9f652ea755ee2a943e7d54171#comments

saeugetier, 2019-04-27:
There is a pull request which adds the "attachInterruptArg" to the core. Currently only "__attachInterruptArg" is available, which may be the same.

https://github.com/esp8266/Arduino/pull/6003/files/c376c97ce1eeaa69a05cf9a12e5e7baf6c366842

saeugetier, 2019-04-27:
`Compiling .pioenvs/esp12e/FrameworkArduino/HardwareSerial.cpp.o
.piolibdeps/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp: In member function 'void SoftwareSerial::enableRx(bool)':
.piolibdeps/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp:139:110: error: 'attachInterruptArg' was not declared in this scope
attachInterruptArg(digitalPinToInterrupt(m_rxPin), reinterpret_cast<void (*)(void*)>(rxRead), this, CHANGE);
^
*** [.pioenvs/esp12e/libe26/EspSoftwareSerial_ID168/SoftwareSerial.cpp.o] Error 1`

dok-net, 2019-04-27:
Sorry about rushing master HEAD, undone now - PR for actually working attachInterruptArg in ESP8266 is pending review.

closed 2019-04-27 by dok-net

farajfarook, 2019-05-10:
What's the solution for this issue?

ddwood, 2019-12-06:
Problem exists for me as well;
PlatformIO in VSCode Home 3.0.1·Core 4.1.0

Contents of platformio.ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

lib_deps =
    13  ; Adafruit_GFX
    135 ; Adafruit_SDD1306 
    EspSoftwareSerial@>=6.4.0
    TMC2208Stepper

I also had an error with IRAM_ATTR not being defined. (Which is not defined in any of the library headers in use).

Any help with this would be greatly appreciated.

dok-net, 2019-12-07:
POBCAK. What leads to updating just one part of the stack to a latest release without also pulling in at least halfway recent versions of the rest?RTFM, don't mix versions manually, report versioning issues to your build system"s vendor. This issue got closed for a reason 🙂