plerup/espsoftwareserial #97: 'attachInterruptArg' was not declared in this scope

farajfarook, 2019-05-10:
https://github.com/plerup/espsoftwareserial/blob/32aa10680170c7a24a85fa076cf390348f370d23/src/SoftwareSerial.cpp#L139

> .piolibdeps/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp: In member function 'void SoftwareSerial::enableRx(bool)':
> .piolibdeps/EspSoftwareSerial_ID168/src/SoftwareSerial.cpp:139:110: error: 'attachInterruptArg' was not declared in this scope

I saw there was another issue related to this which is closed. But the issue is still there when i try to compile it in platformio (Arduino 8285|8266).

carmbrust, 2019-05-15:
I've found that 5.1.0 has this issue. A workaround in PlatformIO: 
add this to your environment: 
lib_deps =
  # ... works with this specific version
  EspSoftwareSerial@5.0.4

dok-net, 2019-05-15:
There is no version 5.1.0 in this repository - HEAD, master, has 5.0.4 as it's most current release. I am sorry, a tooling error caused a push to this repo, I forced a rewind, but the commits are still accessible as your source blob above proves, even if it's not in any actual branch. I don't know what Platformio does to keep using that long gone reverted version. Anyway, real soon now, there will be an ESP8266 Arduino with support for attachInterruptArg and a 5.1.0 EspSoftwareSerial to follow suit, and hopefully everything will cool down after that.
That said, the library manager of the Arduino IDE didn't just pick up a "5.1.0" release, or did it?

farajfarook, 2019-05-22:
I stumble up on this project thinking SONOFF Bridge uses serial over GPIO to communicate with it's RF chip. But ended up realising it's not software but it's actually the hardware serial. 
Thank you for the work you have done so far mate. As a suggestion, if the versions gives you trouble with platformio, I would rather suggest you to update to an upper version number than to revert back the commits to a lower version. We'll never know how the package repos and caching is done in the PIO's end. Cheers!

dok-net, 2019-05-22:
@farajfarook About the versioning, the push to THIS repository on GitHub was completely unintended - what I push to my fork doesn't hurt anyone. The Arduino library manager AFAIK does not publish releases just becauses there's a tag, it relies on creating a release in GitHub that stores a zip archive of the sources. Platformio seems to be broken (possibly intentionally so and they wouldn't call it that way) in this regard.
That being said, yes, I know, force pushing is discouraged for all these predictable reasons :-)
Once ESP8266 Arduino has the patch, I will create a release of EspSoftwareSerial with a version number greater than 5.1.0, that should put this all to rest.

closed 2019-05-22 by dok-net