plerup/espsoftwareserial #99: Compilation error in Arduino v1.8.5, ESP8266 v2.4.0

fenyvesi, 2019-06-02:
I renamed the old SoftwareSerial.h and copied ESPSoftwareSerial v5.0.4 into the ...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries directory.

I got the following compilation error:

> C:\Users\Gyuri\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\EspSoftwareSerial-5.0.4\src\SoftwareSerial.cpp: In function 'void sws_isr() [with int I = 7]':
> 
> C:\Users\Gyuri\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\EspSoftwareSerial-5.0.4\src\SoftwareSerial.cpp:36:38: error: void sws_isr() [with int I = 7] causes a section type conflict with void SoftwareSerial::preciseDelay(uint32_t, bool)
> 
>  template<int I> void ICACHE_RAM_ATTR sws_isr() {
> 
>                                       ^
> 
> C:\Users\Gyuri\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\EspSoftwareSerial-5.0.4\src\SoftwareSerial.cpp:226:22: note: 'void SoftwareSerial::preciseDelay(uint32_t, bool)' was declared here
> 
>  void ICACHE_RAM_ATTR SoftwareSerial::preciseDelay(uint32_t deadline, bool asyn) {
> 
>                       ^

dok-net, 2019-06-03:
@fenyvesi Why would you do such a thing? Anyway, this is an unsupported combination - it was discussed in another issue before. Currently, the version dependencies between EspSoftwareSerial and esp8266 Arduino are quite current in the submodule dependency in git, if you are not using that, please stick to what your IDE offers, if via this way problems occur, please report them here.

closed 2019-06-03 by dok-net

fenyvesi, 2019-06-03:
ESP8266 v2.5.0 works with the built-in version.
Thanks.