plerup/espsoftwareserial #100: SoftwareSerial doesn't read data after a certain time being idle

gutierrezps, 2019-06-12:
I was trying to use a [RDM6300 RFID reader](https://github.com/arduino12/rdm6300) with the ESP8266, when I faced a weird issue: sometimes the ESP wasn't detecting that a tag was being read. Upon adding some debug code, I noticed the `available()` method from SoftwareSerial was returning `false`, even when there was data being sent constantly to the input pin. Upon further tests, I found out this problem always happened at a certain time after the last successful SoftwareSerial reading.

Then I was able to isolate the problem and make it reproducible, as follows.

## Environment

- OS: Windows 7 x64
- IDE: Visual Studio Code + PlatformIO
- PlatformIO core version: 3.6.7
- Espressif 8266 platform version: 2.2.1
- ESP8266 Arduino core version: 2.5.2
- Board: Wemos D1 Mini v2.2.0

## Circuit

[image]

## ESP8266 firmware

```c++
#include <Arduino.h>
#include "SoftwareSerial.h"

const uint8_t pin_softSerialRx = D6;
const uint8_t pin_confirmation = D8;

SoftwareSerial softSerial(pin_softSerialRx, -1);

void setup()
{
    Serial.begin(115200);
    softSerial.begin(9600);

    pinMode(pin_confirmation, OUTPUT);
    digitalWrite(pin_confirmation, LOW);

    Serial.println("\n\nWaiting softSerial input");
}

void loop()
{
    static uint32_t lastSoftSerialInput = 0;
    static uint16_t lastSecondsPrinted = 0;

    if (softSerial.available()) {
        if (millis() - lastSoftSerialInput > 300) Serial.print(" # received: ");

        while (softSerial.available()) {
            Serial.print((char) softSerial.read());
            yield();
        }

        lastSoftSerialInput = millis();
        lastSecondsPrinted = 0;     // restart seconds count on Serial
    }

    // turn LED on for 100ms
    bool ledState = millis() - lastSoftSerialInput < 100;
    digitalWrite(pin_confirmation, ledState);

    uint16_t secondsSinceLastInput = (millis() - lastSoftSerialInput) / 1000;

    if (secondsSinceLastInput != lastSecondsPrinted) {
        lastSecondsPrinted = secondsSinceLastInput;
        if (secondsSinceLastInput == 1) Serial.println();
        Serial.print(secondsSinceLastInput, DEC);
        Serial.print(" ");
    }

    delay(10);
[... 242 lines cut ...]
test no. 242:    timeout        next test in 1000ms
test no. 243:    timeout        next test in 1000ms
...
test no. 265:    timeout        next test in 1000ms
test no. 266:    timeout        next test in 1000ms
test no. 267:    confirmed      next test in 268000ms
test no. 268:    timeout        next test in 1000ms
test no. 269:    confirmed      next test in 270000ms
test no. 270:    confirmed      next test in 271000ms
...
test no. 294:    confirmed      next test in 295000ms
test no. 295:    confirmed      next test in 296000ms
test no. 296:    timeout        next test in 1000ms
test no. 297:    timeout        next test in 1000ms
...
test no. 318:    timeout        next test in 1000ms
test no. 319:    timeout        next test in 1000ms
test no. 320:    confirmed      next test in 321000ms
test no. 321:    timeout        next test in 1000ms
test no. 322:    timeout        next test in 1000ms
test no. 323:    confirmed      next test in 324000ms
test no. 324:    confirmed      next test in 325000ms
```

I replaced the ESP8266, and also tried pins D6 (the main one for this issue), D5, D3 and D4, but the problem still happens. Can someone try to reproduce this issue? I am also open to suggestions to improve this testing procedure.

dok-net, 2019-06-12:
Thank your for this extensive and enlightening issue report. I've superficially verified that there is in indeed a problem in all recent EspSoftwareSerial versions when there has been silence on the RX for just short of 27s, equaling 2^31 cycles at 80MHz ;-) ;-)
I will provide a fix ASAP - naturally for the latest version of EspSoftwareSerial, which unfortunately is pending a minor fix in the ESP8266 Arduino core. I am in active contact with the developers there. So either you'll have to wait for that, or if you are familiar with cobbling together different HEAD versions, you could kindly provide me early feedback for the fix.

gutierrezps, 2019-06-13:
Thank you for the quick reply! Yes, I think I can apply your fix on the ESP8266 core and test SoftwareSerial on top of it. Tell me when you provide the SoftwareSerial fix and I'll test it as soon as I can.

dok-net, 2019-06-13:
OK, I would like to ask you to clone this repo, [](https://github.com/dok-net/Arduino.git) , including all submodules, into your workbook directory .../Arduino/hardware/esp8266com/esp8266 (moving aside the official ESP8266 Arduino tree). Please checkout the master branch - this contains all my pending PRs, so please don't use it for anything but this test, but it simply picks up my latest EspSoftwareSerial in [repo]/libraries/SoftwareSerial, too.

gutierrezps, 2019-06-14:
The fix worked, thanks.

Now, since this is a potentially troublesome issue, do we really have to wait for your pull requests to be merged on ESP8266 core? Is it not possible to cherry pick the fix and release it as a new patch version?

dok-net, 2019-06-14:
Not really, this would bind the same resources that are needed for the full update including the prerequisite PRs, as any release for the ESP8266 now happens via the submodule - at least for GitHub users. I've also done some more fixes for issues that I've found since then, that for the first time allow full-duplex at 38400bps software serial to software serial, in my tests.
And finally, I am not about to open that can of worms, I have just about only enough time for keeping one straight version graph alive. Please use the version from my development repo in the interim.

closed 2019-06-14 by dok-net

gutierrezps, 2019-06-14:
I see. For now, I have already made a workaround when I noticed this issue, and I'm not needing this fix at the moment.

Anyway, I (and maybe others as well) would be grateful if you could add a comment here when the fix gets released.

dok-net, 2019-06-16:
@gutierrezps The SDS011 air quality sensor behaves quite differently from the UARTs in both ESP8266 and ESP32, so I had to do a few tweaks to get that working again - I am sorry, could you please retest with the very latest head from my repo? Thanks!

dok-net, 2019-06-22:
@arduino12 @gutierrezps @fredvik EspSoftwareSerial release 5.2.1 is out now for ESP8266 as integral library, once it's merged by the ESP8266 Arduino team, and immediately for both ESP8266 and ESP32 via all the manual methods that exist from https://github.com/plerup/espsoftwareserial/releases/tag/5.2.1 ;-)
Please mind the breaking change in initalization, the constructor now takes zero arguments, they have been moved to the `begin()` function.