plerup/espsoftwareserial #104: Unable to compile for ESP32 board

sajeel-ahmed, 2019-07-16:
I am getting this error while compiling code for ESP32 board in Arduino IDE. Could you please help me managed out this. I  have tried so many things but all went vain. Thank you for your time. 
 error:
C:\Users\Sajeel\Documents\Arduino\libraries\EspSoftwareSerial-5.2.3\src/SoftwareSerial.h:73:9: error: 'size_t SoftwareSerial::readBytes(char*, size_t)' marked 'override', but does not override

  size_t readBytes(char* buffer, size_t size) override {

         ^

exit status 1
Error compiling for board ESP32 Dev Module.

dok-net, 2019-07-16:
This relies on a change to ESP32 Arduino by me, dating back to April, contained in release tag 1.0.3-rc1
https://github.com/espressif/arduino-esp32/tree/1.0.3-rc1
Please use that version.
I am sorry for any inconvenience, but I also strongly feel that all the improvements are worth the effort to users to keep their versions up-to-date.

sajeel-ahmed, 2019-07-17:
Thanks! but I have checked **"Esp Software Serial"** library in release **1.0.3-rc1** and I found nothing. 
I want to use software serial library for ESP32. Actually I want to interface 5 sensors through serial communication with ESP32 and unfortunately ESP32 has just 3 UART ports available. 

Could you help me in locating specific file you were talking about. Thank you for your time. Thanks alot!

dok-net, 2019-07-18:
The 1.0.3-rc1 applies to the ESP32 Arduino core, not the SoftwareSerial library, and on the ESP32, EspSoftwareSerial is not part of that distribution.
Putting it shortly, download/clone the ESP32 Arduino repository, version 1.0.3-rc1 or later, into your sketchbook's hardware/espressif/esp32 directory. Also download/clone EspSoftwareSerial into libraries.

kennythum, 2019-07-18:
@dok-net sorry sir, im still confused. I download the whole esp32-1.0.3-rc1.zip from [https://github.com/espressif/arduino-esp32/releases/tag/1.0.3-rc1](url) into C:\Documents\Arduino\hardware\espressif\esp32, and downloaded the EspSoftwareSerial into libraries (C:\Documents\Arduino\libraries)....yet the compiler still show error:

>  C:\Documents\Arduino\libraries\espsoftwareserial-master\src/SoftwareSerial.h:71:9: error: 'size_t SoftwareSerial::readBytes(uint8_t*, size_t)' marked 'override', but does not override
> 
>    size_t readBytes(uint8_t* buffer, size_t size) override;
> 
>          ^
> 
> C:\Documents\Arduino\libraries\espsoftwareserial-master\src/SoftwareSerial.h:73:9: error: 'size_t SoftwareSerial::readBytes(char*, size_t)' marked 'override', but does not override
> 
>   size_t readBytes(char* buffer, size_t size) override {
> 
>          ^`
> 
> 

I couldn't figure out any clue anymore  :( , your help is very much appreciated!!

dok-net, 2019-07-18:
@kennythum Did you remember to switch your device ESP32 Dev Module to the new downloaded sources? I am not on the Arduino IDE, but I guess that you have to switch to the C:\Documents\Arduino\hardware\espressif\esp32 one in any case.

kennythum, 2019-07-18:
@dok-net yup, all the contents were there, complete. But still couldnt work :(

kennythum, 2019-07-18:
may i know which compiler are you using??

dok-net, 2019-07-18:
Why? Haven't you set up the ESP32 Arduino environment? Of all things, like so:

```
cd hardware/espressif/esp32/tools
./get.exe
```

closed 2019-07-18 by dok-net

kennythum, 2019-07-18:
yes i did....i even done a clean install, yet still no hope LOL

dok-net, 2019-07-18:
OK, let me see.
Checked out EspSoftwareSerial 5.2.3 (Arduino/libraries/espsoftwareserial)
Checked out Arduino/hardware/espressif/esp32, tag 1.0.3-rc1
Ran get.exe
Build some sample (SDS011 measure.ino [https://github.com/dok-net/esp_sds011.git])

Builds fine. I don't know how to help you beyond this?

kennythum, 2019-07-18:
Alright, i guess i found the root of the problem. 1.0.3-rc1 is not an official version, hence, any pull from github will get you the stable version, which is the 1.0.2. Even i download the 1.0.3-rc1 zip file and copy into Documents/Arduino/hardware/espressif/esp32, the arduino IDE doesn't read it from there. Instead, it read it from my C:\Users\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries!!!

Therefore, I copy and paste the whole folder from 1.0.3-rc1 into that directory, \AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2 and vollah! It compiled successfully!

But i noticed that, by using the 1.0.3-rc1 version, some of my existed library such as HTTPclient compiled error. I guess 1.0.3-rc1 is still not yet fully mature.

Honestly, i'm surprised and confused that why the IDE takes the directory of AppData\Local\Arduino15 instead from the Documents...

dok-net, 2019-07-18:
A git client has no concept of stable releases, just branches and tags. 1.0.3-rc1 is rather old by now, but without knowing details, I'd rather caution you that your HTTPclient compile problems may be due to evolving APIs, not immaturity.
Other than that, IIRC, you can select the packaged platform support OR the ones you install in sketchbook / hardware / … in the IDE.
Anyway, can we consider this here issue solved :-) ?

kennythum, 2019-07-19:
yes, consider it as solved...

kennythum, 2019-07-19:
thank you very much for helping :)

phretor, 2019-09-05:
Platform Espressif 32
--------
Updating espressif32                     @ 1.9.0          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10002.190628 [Up-to-date]

udivankin, 2019-09-11:
Which version of EspSoftwareSerial should I use if I want to stick with ESP32 1.0.2 for now?

Bmooij, 2019-09-13:
> Which version of EspSoftwareSerial should I use if I want to stick with ESP32 1.0.2 for now?

Version [5.0.4](https://github.com/plerup/espsoftwareserial/releases/tag/5.0.4) is the latest version that still works with ESP32 1.0.2

our30K, 2019-09-13:
there is way to modify the current 5.2.9 code base to work with ESP32 1.0.2
see here
https://forum.arduino.cc/index.php?topic=635713.0

ashna-zafar, 2019-12-21:
Hello...can anyone help me please i am beginner with esp32 and i have to create my project because it's urgent......i am trying to compile my sketch.....but one error continuously occurs

Error compiling for board ESP32 Dev Module......

i have tried many different things like i have installed core zip files and did setting by follow this link https://www.esp8266.com/viewtopic.php?f=26&t=14222&start=4
below is my sketch:
#include <Arduino.h>
#include "BluetoothSerial.h"
#include <IRremote.h>
#include <IRac.h>
#include <IRrecv.h>
#include <IRsend.h>
#include <IRtimer.h>
#include <IRutils.h>
#include <ir_NEC.h>
#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run make menuconfig to and enable it
#endif

#define NEC_BITS 32
#define DUTY_CYCLE 1/3 // Carrier duty cycle
#define PERIOD 26 // 38kHz, 1/38kHz =~ 26us
#define ACTIVE_HIGH (PERIOD * DUTY_CYCLE)
#define ACTIVE_LOW (PERIOD - (PERIOD * DUTY_CYCLE))
#define Power 0xFBD22D
#define Up 0xFBE817
#define Down 0xFB38C7

#define TOPBIT 0x80000000
#define NEC_HDR_MARK 9000
#define NEC_HDR_SPACE 4500
#define NEC_BIT_MARK 560
#define NEC_ONE_SPACE 1690
#define NEC_ZERO_SPACE 560
char code;

BluetoothSerial SerialBT;
//int incoming;
char incomingChar;// received value will be stored as CHAR in this variable
const int OutputPin =13;

void setup() {
Serial.begin(9600); //Start Serial monitor in 9600
SerialBT.begin("ESP32test"); //Bluetooth device name
Serial.println("The device started, now you can pair it with bluetooth!");
Serial.println("To Power send: a");//print on serial monitor
Serial.println("To Up send: b");
Serial.println("To Down send: c");
pinMode(OutputPin, OUTPUT);
}
void sendNEC(unsigned long data, int nbits) {
mark(NEC_HDR_MARK);
space(NEC_HDR_SPACE);

for (int i = 0; i < nbits; i++) {
if (data & TOPBIT) {
mark(NEC_BIT_MARK);
space(NEC_ONE_SPACE);
}
else {
mark(NEC_BIT_MARK);
space(NEC_ZERO_SPACE);
}
data <<= 1;
}
mark(NEC_BIT_MARK);
space(NEC_HDR_SPACE);
mark(NEC_BIT_MARK);
space(0);
}
void mark(int time) {
//inverting the signal
digitalWrite(OutputPin, LOW);
delayMicroseconds(time); //digitalwrite removes the modulation to provide a demodulated signal
}
void space(int time) {
digitalWrite(OutputPin, HIGH);
delayMicroseconds(time);
}
void loop() {
incomingChar =(char)SerialBT.read();
if (Serial.available())
{
SerialBT.write(Serial.read());
}
if (SerialBT.available())
{code = SerialBT.read(); //Read what we recevive
Serial.print("Received:");
//Serial.println(receivedChar);//print on serial monitor
//Serial.println(code);
//Serial.println(incoming);
SerialBT.print ("Received:");//print on serial monitor
//SerialBT.println(receivedChar);//print on serial monitor

 if(code == Power)
{Serial.println("Power received");
sendNEC(Power, 32);
Serial.println(code);
if(code == Up )
{digitalWrite(OutputPin, HIGH);
Serial.println("Up received");
sendNEC(Up, 32);}
if(code == Down)
{digitalWrite(OutputPin, LOW);
Serial.println("Down received");
sendNEC(Down, 32);}
}
else
{
sendNEC(Power, 32);
}
delay(20);
}
}

dok-net, 2019-12-21:
@ashna-zafar, I can't see any connection to EspSoftwareSerial in your sketch. I'm sorry.