plerup/espsoftwareserial #107: how to compile with this library in the Windows environment

bill-orange, 2019-07-23:
I am using the Arduino IDE on a Windows machine. How do I install this library so that the compiler picks it over the  software serial in the hardware folders?  I am using the Arduino IDE 1.8.9

dok-net, 2019-07-23:
From `README.md`:

```
The responsible maintainer of the esp8266 repository has kindly shared the
following command line instructions:

To update esp8266/arduino SoftwareSerial submodule to lastest master:

(optional, clean it:)
$ rm -rf libraries/SoftwareSerial
$ git submodule update --init
(now update it:)
$ cd libraries/SoftwareSerial
$ git checkout master
$ git pull

```

Is this the information you are looking for?

dok-net, 2019-07-23:
Oh, and do expect to update ESP8266 Arduino to master, too - at least a fairly recent commit will be required...

bill-orange, 2019-07-24:
I saw the info in your post.  That looks like it applies to a Linux environment rather than a Windows environment.

dok-net, 2019-07-24:
Git bash command line. I am working on Windows, too.

bill-orange, 2019-07-24:
@dok-net Sucess.  Thank You

closed 2019-07-24 by bill-orange