plerup/espsoftwareserial #108: ESp8266 reboot when receiving data

bill-orange, 2019-07-24:
I have a difficult to diagnose issue with  ESP8266 running ESPSoftwareSerial (latest version).  Randomly, in intervals of between 10 minutes and 6 hours the ESp8266 reboots while receiving data at 9600 baud.  At much faster baud rates the problem seems to disappear.

Oddly, there is no debug information to decode.  I see a couple of random characters such at "~fi" and then the reboot.

I know that I am not providing much information.  This is all I have.  I suspect some sort of race condition.

Thoughts?  Troubleshooting suggestions.

dok-net, 2019-07-29:
@bill-orange Are you using ESP8266 Arduino latest master from GitHub, have you tried EspSoftwareSerial 5.2.8 or another version, which one exactly?
If I understand you correctly, 

> random characters such at "~fi"
or

> dev 1153

are on the USB serial debug output? Neither is from EspSoftwareSerial.
I you could provide a [MCVE](https://stackoverflow.com/help/mcve) sketch to showcase the issue, it would be helpful, otherwise, there's no known issue that correlates to your bug. I am sorry in that case.

bill-orange, 2019-07-29:
I just double checked, yes, I am using the ver. 5.2.8. I do have USB serial turned on and it is reporting the dev 1153 error.  That error is apparently associated with some sort of WDT timeout. 

Other folks have reported similar errors but not recently and not often.
example:
https://www.bountysource.com/issues/29875478-softwareserial-wdt-reset

I have continued to "play" with this code and have discovered a few thing.  If I turn off just about all other code ( write to OLED via I2C and WiFi) the problem goes away.  The problem also goes away if I up the baud rate from 9600 (using a different GPS module).  Oddly, increasing processor speed does not increase or decrease the crash frequency.

I am thinking that this is a race condition.

I will try to come up with a minimal example but this is going to be tough.  GPS tend to send data in bursts.  The rate and pattern of the data at 9600 baud could be an issue.  Even with an MCVE sketch, it would likely be necessary to reproduce most of my hardware environment.  Never the less, I will work on it.  Maybe I can write some sort of loop back that will generate a similar software serial load.

What would perhaps be helpful, is some insight on how to troubleshoot this within my hardware environment. What debug print statements, yield () statements or other devices could I add to my code or to EspSoftwareSerial 5.2.8  that might shed some light on this?

Bill

dok-net, 2019-08-27:
@bill-orange Sorry about the delay - if you've not fixed this in your code yet, here's a question:
Could you please check, if the suspected WDT occurs during write()s? Just add some debug message before and after entering EspSoftwareSerial's write functions. Thanks.

dok-net, 2019-08-28:
@bill-orange You got me wondering, what do you mean by "tasks" in this context? Are you using my CoopTasks, or some other library that modifies yield() and/or delay()?
In recent versions of EspSoftwareSerial, I can't imagine how read() is affected by bitrate.
"Scattering yields()" into the write() code would be a very awkward thing to do in the latest releases, are you sure you're looking at the same sources that I am :-) ?
I am myself using the SDS011 particulate matter sensor, at 9600bps on software serial. Btw, are you aware that the ESP8266 can Serial.swap() from USB to GPIOs D7/D8 (IOs 13/15), and you can serve the USB serial via SoftwareSerial then for diagnostics?
Now, about the read() watchdog timeout, have you ever placed yields() just ahead of and after calls to read()? I still can't image read() in current EspSoftwareSerial taking any serious amout of time, but if it's just pushing over the limit because of prior long-running activity without yield(), who knows?

bill-orange, 2019-08-30:
This is getting off topic but I have a question regarding the  SDS011.  The eBay ads show an operating voltage of 5v.  Are your running yours on 5 volts and did you have to do level conversion on the serial output?

dok-net, 2019-08-30:
Per the spec sheet, 5V supply, Rx and Tx at 3.3V :-) I'm running power off the USB serial adapter on the ESP8266 breakout board, Rx/Tx directly connected to the GIOPs. Works as well with ESP32.
https://github.com/dok-net/esp_sds011/releases

bill-orange, 2019-09-17:
Since I recompiled with version 5.3.0 I have not experienced a WDT crash.  Its a bit early to declare victory since the crashes are random.  At this poing, its safe to say that they are considerably less frequent.

nvicchi, 2019-09-23:
Could you give a hint on how to compile latest version? With the version included in esp8266 2.5.2 it happens the same to me (with burst data entry I get a wdt reset). 
But when I download an try to compile 5.3.0, I get the following error:

C:\Users\xxx\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h: At global scope:
C:\Users\xxx\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:145:8: error: expected ';' at end of member declaration
     T& IRAM_ATTR pushpeek()
        ^
C:\Users\xxx\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:145:27: error: ISO C++ forbids declaration of 'pushpeek' with no type [-fpermissive]
     T& IRAM_ATTR pushpeek()
C:\Users\xxx\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:157:10: error: expected ';' at end of member declaration
     bool IRAM_ATTR push();
          ^

The 5.3.0 version installed via the Library Manager, but had to delete the versión included in 2.5.2 to get Arduino to use it.

Thanks!

bill-orange, 2019-09-23:
@nvicchi Wow. I had no issues getting it to compile at all. There must be something different in your environment.  Are you using the latest Arduino IDE?  @dok-net Can you weigh in on this one?

nvicchi, 2019-09-23:
Arduino 1.8.10
NodeMCU 1.0 (ESP-12E Module).
esp8266 by ESP8266 Community (2.5.2)
Had to manually replaced original library with 5.3.0

I just read that dok-net told you to update to masters ESP8266 Community in another issue. I will try that.

bill-orange, 2019-09-23:
I am running Arduino 1.8.9.  I would expect 1.8.10 to be fine.  I am using the ESP 8266 core off the board manager.  It appears to be 2.5.2.  My library updated to 5.3.0 off the Library Manager.  I have no clue what the problem might be.  You could post to the ESP8266 Github site since this is likey to be an issue outside of espsoftwareserial anyway.

You have to be really careful in your posts there as far as completeness and format.  I get my hand slapped frequently.

nvicchi, 2019-09-23:
Bill, thanks a lot. I was actually using 1.8.9 and updated in order to see if that was the issue. 
I updated to 5.3.0 from the Library Manager originally. but it was complaining about having 2 candidates and that it would use the ESP8266 version, ignoring 5.3.0.

I will keep looking into this. If I have to, I will make a clean install in a virtual machine.

By the way, since your last post on the original issue was a week ago, could you confirm it is still working? Did you get any wdt resets?

bill-orange, 2019-09-23:
THE GPS is still running fine.  The trouble with an intermittent is that you never know when it is really fixed!

I am now also  running an SDS011 sensor (like @dok-net ) on a Huzzah Feather ESP8266 without any problems.  Provisionally , things are looking good.

Good idea to test in the VM.

nvicchi, 2019-09-23:
Statistically, its a huge improvement. In my case, the error is 100% repeatable. I know exactly how to trigger it. So when and if I can compile the 5.3.0 version, I will let you know if its fixed or not.

kross222, 2019-09-24:
Arduino 1.8.10
ESP-12E Module
esp8266 by ESP8266 Community (2.5.2)
original library 5.3.0

not compiled:

In file included from C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/SoftwareSerial.h:27:0,

           from C:\Users\non\Documents\Arduino\sketch_sep24a\sketch_sep24a.ino:5:

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:145:8: error: expected ';' at end of member declaration

     T& IRAM_ATTR pushpeek()

        ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:145:27: error: ISO C++ forbids declaration of 'pushpeek' with no type [-fpermissive]

     T& IRAM_ATTR pushpeek()

                           ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:157:10: error: expected ';' at end of member declaration

     bool IRAM_ATTR push();

          ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:157:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:145:8: note: previous declaration 'T& circular_queue<T>::IRAM_ATTR'

     T& IRAM_ATTR pushpeek()

        ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:157:25: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]

     bool IRAM_ATTR push();

                         ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:164:10: error: expected ';' at end of member declaration

     bool IRAM_ATTR push(T&& val);

          ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:164:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:145:8: note: previous declaration 'T& circular_queue<T>::IRAM_ATTR'

     T& IRAM_ATTR pushpeek()

        ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:164:32: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]

     bool IRAM_ATTR push(T&& val);

                                ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:171:10: error: expected ';' at end of member declaration

     bool IRAM_ATTR push(const T& val)

          ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:171:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:145:8: note: previous declaration 'T& circular_queue<T>::IRAM_ATTR'

     T& IRAM_ATTR pushpeek()

        ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:171:37: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]

     bool IRAM_ATTR push(const T& val)

                                     ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:253:16: error: expected initializer before 'circular_queue'

 bool IRAM_ATTR circular_queue<T>::push()

                ^

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src/circular_queue/circular_queue.h:268:16: error: expected initializer before 'circular_queue'

 bool IRAM_ATTR circular_queue<T>::push(T&& val)

                ^
				
Several libraries found for "SoftwareSerial.h"
  Use: C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial
exit status 1
Compilation error for the Generic ESP8266 Module board.

kross222, 2019-09-24:
i choose "generic esp8266"

nvicchi, 2019-09-24:
That last phrase was for bill-orange :)
Did you try modifying 
`#if !defined(ESP32) && !defined(ESP8266)
#define ICACHE_RAM_ATTR
#define IRAM_ATTR
#endif`

to

`#if !defined(ESP32) && !defined(ESP8266)
#define ICACHE_RAM_ATTR
#define IRAM_ATTR
#endif
#define IRAM_ATTR`

bill-orange, 2019-09-24:
@nvicchi @kross222 It's a mystery to me why I can compile.  I have to assume that some other fix for some other reason fixed this too.

That being said, take a look at closed issues 112 and 107 in this git.

kross222, 2019-09-24:
modifying it. So...
C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src\SoftwareSerial.cpp: In member function 'void SoftwareSerial::enableRx(bool)':

C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial\src\SoftwareSerial.cpp:140:121: error: 'attachInterruptArg' was not declared in this scope

             attachInterruptArg(digitalPinToInterrupt(m_rxPin), reinterpret_cast<void (*)(void*)>(rxBitISR), this, CHANGE);
                                                                                                                        
Несколько библиотек найдено для "SoftwareSerial.h"
 Используется: C:\Users\non\Documents\Arduino\libraries\EspSoftwareSerial
exit status 1
Ошибка компиляции для платы Generic ESP8266 Module.

nvicchi, 2019-09-24:
Success! Just tried it in a fresh Arduino install in Mac.
Install Arduino to Applications folder, updated esp8266 from github (just followed the readme instructions), then updated espsoftwareserial as in 107 issue.
Only tried to compile the test example, but worked!

Thanks bill! 

Kross, follow those instructions and you should be fine!

bill-orange, 2019-09-25:
@nvicchi  Excellent news! Now we can see if the new version resolves your wdt problem.  Keep us posted.

nvicchi, 2019-09-25:
In fact it did! It doesn't fail anymore. 
My test case was simple. If a burst of more than a certain amount of bytes (not sure how many, about 150 I might guess) arrived, it would immediately fail. If the number of bytes is of importance I can check. Now it doesn't fail anymore.

I still can make it fail if, at the same time that the burst arrives, I access a web server running in parallel. I don't care actually a lot about this situation right now.

By the way, I could succesfully compile it in windows too. It was harder (the doc states "the arduino directory" which pretty straight forward for linux and Mac, but it is not for Windows. You have /user/AppData location, Program files location, Documents locations. Tried several times and eventually worked in the Documents/Arduino location.

bill-orange, 2019-09-25:
Excellent!  I will close this.  @kross222 if you are still having problems it would be best to open a new issue.  The title of this issue really does not reflect your problem.

closed 2019-09-25 by bill-orange