plerup/espsoftwareserial #112: Does not compile

likengmx, 2019-08-21:
THIS IS THE ERROR:

circular_queue is headache
_________________________________________________________________
Arduino: 1.8.9 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from /home/ni/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/SoftwareSerial.h:27:0,
                 from /home/ni/Arduino/BatAn11/BatAn11.ino:263:
/home/ni/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:144:10: error: expected ';' at end of member declaration
     bool IRAM_ATTR push(T&& val);
          ^
/home/ni/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:144:32: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]
     bool IRAM_ATTR push(T&& val);
                                ^
/home/ni/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:151:10: error: expected ';' at end of member declaration
     bool IRAM_ATTR push(const T& val)
          ^
/home/ni/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:151:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'
/home/ni/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:144:10: note: previous declaration 'bool circular_queue<T>::IRAM_ATTR'
     bool IRAM_ATTR push(T&& val);
          ^
/home/ni/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:151:37: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]
     bool IRAM_ATTR push(const T& val)
                                     ^
/home/ni/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:216:16: error: expected initializer before 'circular_queue'
 bool IRAM_ATTR circular_queue<T>::push(T&& val)
                ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

dok-net, 2019-08-27:
I'll need to set up a pristine Arduino environment in a virtual machine to check this. In the meantime, I'm convinced that it's just a malconfiguration on your side - stick with me, I'll let you know what I find out using the Arduino IDE instead of my build environment.
If you've resolved the problem in the meantime, please let me know and help me save the time :-)

kbickham, 2019-08-31:
I have same errors, using the nightly build.

dok-net, 2019-09-01:
RTFM; basically, use the packaged ESP8266 BSP without downloading EspSoftwareSerial separately (the path hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/SoftwareSerial.h hints that it's included).
Or use ESP8266 Arduino community via Git download and follow the instruction in EspSoftwareSerial's README to update the (again) included EspSoftwareSerial. Any separate EspSoftwareSerial in Arduino/libraries will interfere.
Please don't try to update EspSoftwareSerial without going to ESP8266 Arduino from Git master as well.

closed 2019-09-01 by dok-net