plerup/espsoftwareserial #113: ESP.getcyclecount() instruction in ISR causes fatal error: cache disabled but cached memory region accessed on ESP32

ianwillianb, 2019-09-01:
I found out that even though the ISR is store in IRAM, by default the instruction ESP.getcyclecount() is stored in flash, so if the access to flash is temporary disabled and an interrupt of type rxbitsISR() occurs the ESP32 ends up crashing. 
I understand this is an issue out of the scope of the library, but this fact doesn't make it stable without adding IRAM_ATTR to the function ESP.getcyclecount() on ESP32. 
Is there a way to change the way the ISR gets the number of cycles of the CPU, without compromising performance and functionality of the library? #

dok-net, 2019-09-01:
One could duplicate the inline assembly instructions into the ISR handler... but no, thanks ;-) I'll try to get PR accepted with the ESP32 Arduino team - keep your fingers crossed and wish me good luck with that.
Otherwise, a great many thanks for pointing this out, while I've never seen a crash on the ESP32 myself.
Allow me a question: have you actually done what you said, adding the IRAM_ATTR, and resolved those crashes, or is it a theoretical fix at this time?

ianwillianb, 2019-09-01:
Yes, adding the IRAM_ATTR solves the crashes. It is running stable for over 24 hours now, before adding this modification crashes occurred in the the first few minutes of execution. 
Doing operations on NVS (only at startup, before calling SoftwareSerial::begin() ), or even having multiples tasks running on both cores could be the cause of the issue in my case, since you've never experience a crash of this kind.
I have to thank you guys for providing this wonderful library :)

dok-net, 2019-09-01:
@ianwillianb Could you try my approach in PR https://github.com/espressif/arduino-esp32/pull/3165, which is to inline the machine code in question instead placing the function in IRAM?