plerup/espsoftwareserial #115: ESP32 wrover

vangalvin, 2019-09-24:
I seem to be getting these errors when compiling swsertest on a wrover ESP32

In file included from E:\Arduino\libraries\espsoftwareserial-master\examples\swsertest\swsertest.ino:5:0:

E:\Arduino\libraries\espsoftwareserial-master\src/SoftwareSerial.h:70:12: error: 'size_t SoftwareSerial::readBytes(uint8_t*, size_t)' marked 'override', but does not override

     size_t readBytes(uint8_t* buffer, size_t size) override;

            ^

E:\Arduino\libraries\espsoftwareserial-master\src/SoftwareSerial.h:72:12: error: 'size_t SoftwareSerial::readBytes(char*, size_t)' marked 'override', but does not override

     size_t readBytes(char* buffer, size_t size) override {

            ^

exit status 1
Error compiling for board ESP32 Wrover Module.

Trottero, 2019-09-24:
Is this the same issue as in #104 ?

vangalvin, 2019-09-24:
Could be yes.. I am using 1.0.2 mainly for compatibility with the Wireless Shell board i have.

Trottero, 2019-09-24:
I had the same issue as you and upgrading to the dev build fixed it for me. IIRC there were a couple of other options presented in the issue, you could give those a try :)

vangalvin, 2019-09-24:
Awesome, I will give that a go. it may also solve the other issue I am having. 
Using the hardware serial the RockBLOCK sat modem is causing the ESP32 to cyclic reboot hence why I was going to try using Softwareserial instead.

Trottero, 2019-09-24:
I don't have any experience with any sat modems but according to this stack exchange post (https://arduino.stackexchange.com/questions/34126/whats-the-difference-between-all-the-software-serial-libraries-which-one-is-ar ) hardware serial is almost always the better option, your issue might be caused by something else

vangalvin, 2019-09-24:
> hardware serial is almost always the better option, your issue might be caused by something else

Yea, I would have preferred to use hardware serial however I suspect there is something weird going on with the ESP32 hardware side that is preventing it from running the code. and given the deadline is only a couple of weeks away I am getting a little desperate for a solution.

dok-net, 2019-09-27:
Arduino ESP32 1.0.3 is currently the latest release tag - at least EspSoftwareSerial 5.3.0 is known to work with that. At the time of writing this, the relevant patch to Arduino ESP32 to align with ESP8266, in order to let EspSoftwareSerial compile, is 5 months old.
Before entering issue reports, please prepare a clean build environment, check the READMEs, and install the latest packages accordingly, If you really think you need to combine older packages, please check which version of BSPs and libraries were released to work together, one hint of that would be the release time stamps.
You can generally expect little interest on the side of library developers to backport or "fix" phantom issues due to mismatched versions - remember that this is a volunteer effort, for free.

dok-net, 2019-09-27:
In closing, I would like to add, that the purpose of the issue tracking for this project is to report, in reproduceable detail, problems in the software to the developer(s). There are other sites on the internet for peer to peer user support, please consider that these ongoing discussion clutter my inbox and thus reduce my interest, in the worst case, causing me to miss reports of real bugs.

closed 2019-09-27 by dok-net