plerup/espsoftwareserial #116: Error in esp8266 and platformio

psinetron, 2019-09-24:
Hi!. I don't understand but library not working witk platformio
I use:
platform io
esp8266
software serial is 5.3.0
and I have an error:
`In file included from .pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/SoftwareSerial.h:27:0,
                 from .pio\libdeps\esp12e\EspSoftwareSerial_ID168\src\SoftwareSerial.cpp:23:
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:145:8: error: expected ';' at end of member declaration
     T& IRAM_ATTR pushpeek()
        ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:145:27: error: ISO C++ forbids declaration of 'pushpeek' with no type [-fpermissive]
     T& IRAM_ATTR pushpeek()
                           ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:157:10: error: expected ';' at end of member declaration
     bool IRAM_ATTR push();
          ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:157:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:145:8: note: previous declaration 'T& circular_queue<T>::IRAM_ATTR'
     T& IRAM_ATTR pushpeek()
        ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:157:25: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]
     bool IRAM_ATTR push();
                         ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:164:10: error: expected ';' at end of member declaration
     bool IRAM_ATTR push(T&& val);
          ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:164:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:145:8: note: previous declaration 'T& circular_queue<T>::IRAM_ATTR'
     T& IRAM_ATTR pushpeek()
        ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:164:32: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]
     bool IRAM_ATTR push(T&& val);
                                ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:171:10: error: expected ';' at end of member declaration
     bool IRAM_ATTR push(const T& val)
          ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:171:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:145:8: note: previous declaration 'T& circular_queue<T>::IRAM_ATTR'
     T& IRAM_ATTR pushpeek()
        ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:171:37: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]
     bool IRAM_ATTR push(const T& val)
                                     ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:253:16: error: expected initializer before 'circular_queue'
 bool IRAM_ATTR circular_queue<T>::push()
                ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:268:16: error: expected initializer before 'circular_queue'
 bool IRAM_ATTR circular_queue<T>::push(T&& val)
                ^
In file included from .pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/SoftwareSerial.h:27:0,
                 from src\main.cpp:6:
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:145:8: error: expected ';' at end of member declaration
     T& IRAM_ATTR pushpeek()
        ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:145:27: error: ISO C++ forbids declaration of 'pushpeek' with no type [-fpermissive]
     T& IRAM_ATTR pushpeek()
                           ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:157:10: error: expected ';' at end of member declaration
     bool IRAM_ATTR push();
          ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:157:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:145:8: note: previous declaration 'T& circular_queue<T>::IRAM_ATTR'
     T& IRAM_ATTR pushpeek()
        ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:157:25: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]
     bool IRAM_ATTR push();
                         ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:164:10: error: expected ';' at end of member declaration
     bool IRAM_ATTR push(T&& val);
[... 37 lines cut ...]
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src\SoftwareSerial.cpp:60:170:   required from here
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/circular_queue/circular_queue.h:62:5: error: uninitialized reference member 'circular_queue<unsigned int>::IRAM_ATTR' [-fpermissive]
src\main.cpp:96:33: error: no matching function for call to 'SoftwareSerial::SoftwareSerial(int, int, bool, int)'
 SoftwareSerial ss(5,10,false,128);
                                 ^
src\main.cpp:96:33: note: candidate is:
In file included from src\main.cpp:6:0:
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/SoftwareSerial.h:47:5: note: SoftwareSerial::SoftwareSerial()
     SoftwareSerial();
     ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/SoftwareSerial.h:47:5: note:   candidate expects 0 arguments, 4 provided
src\main.cpp: In function 'void setup()':
src\main.cpp:101:16: error: no matching function for call to 'SoftwareSerial::begin(int)'
   ss.begin(9600);
                ^
src\main.cpp:101:16: note: candidate is:
In file included from src\main.cpp:6:0:
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/SoftwareSerial.h:51:10: note: void SoftwareSerial::begin(uint32_t, int8_t, int8_t, SoftwareSerialConfig, bool, int, int)
     void begin(uint32_t baud, int8_t rxPin, int8_t txPin = -1,
          ^
.pio\libdeps\esp12e\EspSoftwareSerial_ID168\src/SoftwareSerial.h:51:10: note:   candidate expects 7 arguments, 1 provided
*** [.pio\build\esp12e\src\main.cpp.o] Error 1
*** [.pio\build\esp12e\lib435\EspSoftwareSerial_ID168\SoftwareSerial.cpp.o] Error 1`

I have seen issue #103 but it not help me

nvicchi, 2019-09-25:
Check ESp8266 reboot when receiving data #108

kross222, 2019-09-26:
@psinetron remove ESP8266 and espSoftwareSerial and then install again.

psinetron, 2019-09-26:
reinstall not helped.
It work on ArduinoIDE but not working on PlatformIO for visual studio. Why?

nvicchi, 2019-09-26:
Did you checked #108? Had all the same problems. You must update ESP8266 toolchain directly from github and after that do the same with the espsoftwareserial to version 5.3.0.
With the ESP8266 2.5.2 wont work out of the box. There are dependencies that are not fitted.

psinetron, 2019-09-26:
Ok. I set latest ESP8266 library and 5.3.0 softwareserial
```
[env:esp12e]
platform = https://github.com/platformio/platform-espressif8266.git
board = esp12e
framework = arduino

lib_deps =
     # Using library Id
     168

     # Using library Name
     EspSoftwareSerial
     EspSoftwareSerial@>=5.3.0

```
Whats wrong?

psinetron, 2019-09-27:
I changed esp8266 platform to 1.8.0 (It Arduino core to 2.4.2) It not working again. Help me guys.
I use windows.
platformio.ini for example:
```
[env:esp12e]
platform = espressif8266@1.8.0
board = esp12e
framework = arduino
lib_deps =
     # Using library Id
     168

     # Using library Name
     EspSoftwareSerial@5.3.0
```

dok-net, 2019-09-27:
Please look at [Travis CI build](https://travis-ci.org/esp8266/Arduino/builds/585344207) of  of merging PR https://github.com/esp8266/Arduino/pull/6526 . You'll see that the PlatformIO builds succeeded.

Building some software versions "1.8.0" and "2.4.2" with EspSoftwareSerial 5.3.0 is not a supported combination by my understanding. What you are doing, and what the compiler messages indicate, is a complete mess.

Please check my comments in a similar issue, to the same effect: https://github.com/plerup/espsoftwareserial/issues/115#issuecomment-536112899

I personally don't use Platformio, never have, don't expect a need to do so in the near future, my hunch is that it uses the same binary package of the Arduino ESP8266 BSPs software as the Arduino board manager, and as such, the correct matching EspSoftwareSerial library version is already included, you should not have a need to install any version separately, in fact, this may be rather harmful. Check the README file of EspSoftwareSerial on how to get updated versions of the software.

closed 2019-09-27 by dok-net