plerup/espsoftwareserial #119: Speed softwareserial for Dutch P1 readings

stevenveenma, 2019-11-09:
Recently I have tested a software repository that uses the softwareserial library https://github.com/neographikal/P1-Meter-ESP8266-MQTT to read the Dutch P1 meter and send the results by using MQTT. I indeed receive correct Telegrams in the right format, but the interval is very intermittent. Sometimes the interval is various minutes instead of the usual 10 seconds. 

I am told that the softwareserial library is not capable of processing the Telegrams with the right speed. But the softwareserial library seems to be capable of handling the P1 baudrate. So I am wondering if the rootcause indeed is this library or that something else is wrong.  

A workaround might be to stop receiving messages as long as processing of the preceding message is running. I rather have a longer but stable interval between the telegrams than the current intermittent behaviour. What are your thoughts about this?

dok-net, 2019-11-09:
@stevenveenma You are not using a recent release of EspSoftwareSerial:
```
mySerial(SERIAL_RX, -1, true, MAXLINELENGTH);
```
This doesn't compile for quite a while now.

:-)

dok-net, 2019-11-09:
And please do report your findings, real HW testing of the latest 115200bps receive code is highly appreciated.

dok-net, 2019-11-09:
See [here](https://github.com/esp8266/Arduino/issues/6737#issuecomment-552135710) for a discussion of the changes in the API.
Wrt. "stop receiving messages as long as processing of the preceding message is running", you should have your buffering straight.
I feel I have to enforce the MCVE rule more strictly, especially on or shortly after releases of the BSP.

closed 2019-11-09 by dok-net