plerup/espsoftwareserial #120: New 2.6.0 core, problem Software Serial causes exception when initialized with large buffer (2k buffer)

egzola, 2019-11-09:
New 2.6.0 core, problem Software Serial causes exception when initialized with large buffer (2k buffer)

Example:

softser.begin(57600, rx, tx, false, 2048);

In 2.5.0 core it works fine...

dok-net, 2019-11-09:
To quote:
@egzola please report at the espsoftwareserial repo, not here. Also, please provide an MCVE as well as context for the usage. The info you provided here is not nearly enough to consider.
Closing in 5...

dok-net, 2019-11-09:
Please consider this: 2048 byte buffer size seems unreasonable to begin with. But: this results in about 40 times as much additional internal bit receive buffer. Without your sketch or library this most likely cannot be reproduced. With it, on the other hand, there's nothing for the developers here to do. Check with your sketch or library authors to reduce the buffer size.

dok-net, 2019-11-09:
ESP8266 core 2.5.0 referenced a version of EspSoftwareSerial that's a completely different beast. I doubt it works on the latest ESP8266 core release, due to changed timings in the core BSP.
The internal buffers in EspSoftwareSerial should not be used to buffer large datagrams, please read as often as possible into user provided buffers, and reduce the size of the SW serial buffer accordingly.

closed 2019-11-09 by dok-net

dok-net, 2019-11-10:
@egzola EspSoftwareSerial 6.0.0 has backward compatibility now for the common ctor/begin() setup. If default arguments are overridden, please adapt to the new API.
Please either wait for ESP8266 Arduino core 2.6.1 or use Git master, following instructions on how to update to the HEAD revision of the EspSoftwareSerial submodule (in libraries/SoftwareSerial)