plerup/espsoftwareserial #122: does not compile

fotoamg, 2019-11-10:
Hi, I put the latest now and the circular_queue.h does not compile with my Arduino IDE :(

In file included from C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/SoftwareSerial.h:27:0,

                 from C:\wpand\Nextion\pip2000custom1\pipboySWserial1_nextion\pipboySWserial1_nextion.ino:7:

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:145:8: error: expected ';' at end of member declaration

     T& IRAM_ATTR pushpeek()

        ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:145:27: error: ISO C++ forbids declaration of 'pushpeek' with no type [-fpermissive]

     T& IRAM_ATTR pushpeek()

                           ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:157:10: error: expected ';' at end of member declaration

     bool IRAM_ATTR push();

          ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:157:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:145:8: note: previous declaration 'T& circular_queue<T>::IRAM_ATTR'

     T& IRAM_ATTR pushpeek()

        ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:157:25: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]

     bool IRAM_ATTR push();

                         ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:164:10: error: expected ';' at end of member declaration

     bool IRAM_ATTR push(T&& val);

          ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:164:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:145:8: note: previous declaration 'T& circular_queue<T>::IRAM_ATTR'

     T& IRAM_ATTR pushpeek()

        ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:164:32: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]

     bool IRAM_ATTR push(T&& val);

                                ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:171:10: error: expected ';' at end of member declaration

     bool IRAM_ATTR push(const T& val)

          ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:171:10: error: redeclaration of 'bool circular_queue<T>::IRAM_ATTR'

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:145:8: note: previous declaration 'T& circular_queue<T>::IRAM_ATTR'

     T& IRAM_ATTR pushpeek()

        ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:171:37: error: ISO C++ forbids declaration of 'push' with no type [-fpermissive]

     bool IRAM_ATTR push(const T& val)

                                     ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:253:16: error: expected initializer before 'circular_queue'

 bool IRAM_ATTR circular_queue<T>::push()

                ^

C:\Users\gamma10\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/circular_queue/circular_queue.h:268:16: error: expected initializer before 'circular_queue'

 bool IRAM_ATTR circular_queue<T>::push(T&& val)

                ^

dok-net, 2019-11-10:
New EspSoftwareSerial depend on current ESP8266 Arduino releases. Also, please read the docs!

dok-net, 2019-11-10:
Using the library manager is not just unnecessary - it's unsupported. This as additional info to the documentation.

closed 2019-11-10 by dok-net