plerup/espsoftwareserial #123: Removal of the (receivePin, transmitPin, inverse_logic, bufSize, isrBufSize) constructor after 5.0.4

pawelsky, 2019-11-16:
Hi,

first of all - great library. Glad to see an actively maintained SoftSerial for ESP platforms.

Quick question though - any particular reason for which the
`SoftwareSerial(int receivePin, int transmitPin, bool inverse_logic = false, int bufSize = 64, int isrBufSize = 0);`
constructor has been removed and later replaced in 6.0.0 with just:
`SoftwareSerial(int8_t rxPin, int8_t txPin = -1);`

That breaks compatibility with the standard Arduino SoftSerial library constructor
`SoftwareSerial(uint8_t receivePin, uint8_t transmitPin, bool inverse_logic = false);`
and makes it more difficult to maintain multi-platform libraries using inverted softserial.

Would you consider adding it back?

dok-net, 2019-11-16:
I'll check. Either there is a compatibility reason, hardware serial on ESP8266 and ESP32 is what's aimed for, or I quite frankly goofed and didn't notice that inverse_logic via default parameter can be included in the compatibility constructor.

dok-net, 2019-11-16:
@pawelsky Could you please refer to some/any 3rd party libraries that actually internally initialize any SoftwareSerial incarnation including the inversion flag? I'd here discount example sketches that include that parameter just for void reasons, and ask for issue reports against these instead.
The background is, that the duality constructor and begin makes it impossible to tell if the begin() call defaults the inverse argument, leaving any in the constructor call active, or if it overrides it to false. Reengineering this is really just additional needless code where if all else fails an ``#ifdef ESP8266`` is on order. I suspect there's more differences in the API and the capabilities and pin settings anyway, that ask for this, so the ctor and begin() syntax is just one in the crowd.

pawelsky, 2019-11-17:
@dok-net Sure, here are 2 examples of my library for FrSky telemetry
https://www.rcgroups.com/forums/showthread.php?2245978-FrSky-S-Port-telemetry-library-easy-to-use-and-configurable
https://www.rcgroups.com/forums/showthread.php?2465555-FrSky-(old)-telemetry-library-easy-to-use-and-configurable
they both initialize the soft serial by calling something like:
```
  softSerial = new SoftwareSerial(pin, pin, true);
  softSerial->begin(speed);
```
I wanted to make them available to as many popular platforms as possible, but lack of the backwards compatibility with AVR's and ESP8366's SoftSerial (which still is distributed with 5.0.4 version of the library probably for the same reason) kind of discourages me from that.

An yes, I understand the ambiguity here, but I believe the logic for the invert flag shall be the same as for tx/rx pins - if not specified explicitly, value set by the constructor shall be used.

The only thing I don't like about this solution is that to achieve it the invert flag type would have to be changed from bool to int8_t. Not clean, but it would do the job.

dok-net, 2019-11-18:
@pawelsky I’ve refactored the ctor and begin() to match the call syntax you describe, while staying compatible to the latest changes.
Could you please verify from my development repositories [master branch](https://github.com/dok-net/espsoftwareserial.git) that this suits your requirements, before I push a new release?

pawelsky, 2019-11-18:
Thanks a lot - will test it tonight and report back. 

One question - wouldn't it be safer if we make the txpin and invert parameters non-optional in begin? That way one will not accidentally overwrite invert flag to false when calling begin with just the rx pin specified.

dok-net, 2019-11-19:
I believe I've a well-thought out set of ctor() and being() overloads now for max compatibility - if you can show how the most similar API to all of ESP8266, ESP32, AVR Arduino that I've come up with so far can be further improved, go ahead, I'm all for it. Please don't make me spend too much time on research myself though, I don't know how "accidentally" applies to source code writing ;-)

dok-net, 2019-11-19:
ESP32 `HardwareSerial.h`:
```
void begin(unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false, unsigned long timeout_ms = 20000UL);
```
It's a nod to software serial that one must specify an rxPin, no default, and thus there's no more SoftwareSerialConfig default possible, but beyond that, it's fine using the defaults.

dok-net, 2019-11-19:
@devyte FYI

pawelsky, 2019-11-19:
> but beyond that, it's fine using the defaults.

There is one problem though - when you use one begin variant with just non-default parameters provided the invert flag set in constructor will be kept, but if you use the other one it will be overwritten with false. I'm afraid this inconsistency may lead to some head-scratching if someone is not aware of that. Other than that it is fine.

dok-net, 2019-11-19:
@pawelsky The latest drop in my development repo should have you covered - thanks for the feedback. Could just please sign off on it, if it works for you?

pawelsky, 2019-11-19:
@dok-net thanks, I'll test it tonight and let you know.

pawelsky, 2019-11-19:
@dok-net Did the test, API seems fine, but I've found one issue with RX pin setup. When setting the RX pin mode to input, it is always pulled up (to signal the IDLE state). This is fine for non-inverted serial, but for inverted serial it should actually be pulled down. To fix this both instances of
`pinMode(m_rxPin, INPUT_PULLUP);`
shall be replaced with something like
`pinMode(m_rxPin, m_invert ? INPUT_PULLDOWN : INPUT_PULLUP);`

dok-net, 2019-11-20:
@pawelsky Thanks for the feedback. I am sorry but I have to report that I will not agree to the latest change you propose. The implementation of pullup or pulldown internal resistors differs between ESP32, ESP8266, and is even dependent on the exact GPIO used. Except GPIO16 on ESP8266, it is my take that INPUT_PULLUP is the only selection in the API, which is also generally matched by the HW.
As far as the receiving code is concerned, while there may be a spurious detection of level edge on connect, if the resting level disagrees with line signaling of idle/stop, this is a minor nuisance in my book - the important part that there are no continuing random receives on an open connection is achieved nonetheless.

closed 2019-11-20 by dok-net

dok-net, 2019-11-20:
I'd like to add that GPIO16 is historically not available for RX, only TX, in EspSoftwareSerial, here the compatibility issue conveniently outweights other considerations :-)

pawelsky, 2019-11-20:
@dok-net you can probably use just INPUT instead of INPUT_PULLDOWN, but on ESP32 INPUT_PULLUP on inverted. single wire serial will cause the receive to not work. I'm afraid the change necessary.

dok-net, 2019-11-20:
@pawelsky You can expect me to have tested the same (only ESP32 having invert in the HW serial API) and mine doesn't match your finding. hem.

dok-net, 2019-11-20:
@pawelsky 

> inverted, single wire

Is there anything special about single wire use?

dok-net, 2019-11-20:
Please run from release 6.2.1, on ESP32, the example loopback.ino without changes (except inversion in lines 100 and 121). This works fine on my ESP32 board.

pawelsky, 2019-11-20:
@dok-net I've been testing single-wire, inverted 57600 communication with external device connected at pin 4 when I saw the issue, however I'm not able to reproduce the issue anymore today. Could have been some wiring or timing issue. We can call the issue closed for now. Sorry for the trouble, and thanks for pushing the 6.2.2 version.