plerup/espsoftwareserial #125: SIGSEGV, Segmentation fault in circular_queue.h:336

mathertel, 2019-11-27:
I get "Segmentation fault" errors in line #336 of circular_queue.h.
avail : 1073690452
size : 1073690452
Compiling for esp8266 using the built-in and the master version of the library.,

Program received signal SIGSEGV, Segmentation fault.
circular_queue<unsigned char>::pop_n (this=0xd, buffer=<optimized out>, size=1073690452) at C:\Users\Matthias\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\SoftwareSerial\src\circular_queue/circular_queue.h:336
336	    size_t n = min(avail, static_cast<size_t>(m_bufSize - outPos));
Execute debugger commands using "-exec <command>", for example "-exec info registers" will list registers in use (when GDB is the debugger)
Value 'variables' is not valid for field of type 'ContextValue'.
Value 'variables' is not valid for field of type 'ContextValue'.

call stack seams to be corrupted.

dok-net, 2019-11-27:
Here's the usual call for an MCVE.
Besides, you're right,
``circular_queue::pop_n (this=0xd, buffer=, size=1073690452) ``
is pretty corrupted, `0xd` for the `this` pointer is unlikely and 1GB buffer size on an ESP8266 :-)
This has near 100% nothing to do with EspSoftwareSerial.

mathertel, 2019-11-27:
stack gets corrupted in calling available() after reading the buffer empty and processing the received string (that takes some time). I try to give you more details but debugging is not easy in esp8266...

dok-net, 2019-11-27:
You'll really have to share the sequence of actions that cause this in the form of a standalone sketch. Sent from my Huawei Mobile

mathertel, 2019-11-30:
It is still in a huge project.
I'll investigate next week.
You can close this request for now - I'll come back in case I find the root cause.

dok-net, 2019-11-30:
@mathertel To make sure, I've tried to overflow the buffers, and while there was a report missing when the secondary byte buffer overflows (added in my dev repo now), it nonetheless didn't cause any crashes. So yes please, try to find out what the exact cause is, if it's something to do with EspSoftwareSerial, I'll be glad to fix it.

closed 2019-11-30 by dok-net