plerup/espsoftwareserial #127: Use timedRead's for readBytes()

dirkmueller, 2019-12-15:
SoftwareSerial inherits from Stream, which has this virtual function:

```
// read characters from stream into buffer
// terminates if length characters have been read, or timeout (see setTimeout)
// returns the number of characters placed in the buffer
// the buffer is NOT null terminated.
size_t Stream::readBytes(char *buffer, size_t length) {
    size_t count = 0;
    while(count < length) {
        int c = timedRead();
        if(c < 0)
            break;
        *buffer++ = (char) c;
        count++;
    }
    return count;
}
```

unfortunately this is being overwritten by softwareSerial with a variant that never blocks. While that is documented that way, I fall into that trap because I assumed the SoftwareSerial behavior would reflect the Stream behavior (blocking until timeout).

Is there a way to align this or at least have a way to do (blocking) reads of fixed number of bytes? I realize that find and findUntil() are variants that do block, its just that readBytes is the odd one out.

dok-net, 2019-12-16:
I'm torn. If you've looked into the sources, it's obvious that readBytes was overridden because rx data is already in a buffer and copying out single bytes one by one is too inefficient for my liking. `Available()` currently does busy-loop polling already, but I'd like to move away from that to an event based async mode. Therefore, it's rather not on my list to base blocking APIs on delay and repeat, but I'd like to facilitated the work I am doing on Delegates, the unified CONT task suspension and re-scheduling from ISR. To get a handle on the feasibility of that, in order not to build something that hurts overall performance but rather improves latency instead, there is a need to do some performance measurements.
By the way, I have noticed that the comments in Stream are incorrect, they mention private access while the pertaining parts are protected. Also, the use of millis() is really wasting a lot of CPU cycles.
In short, Stream.h|cpp should really be fixed - hint: probably using PolledTimeout.

Enough said, in order to provide a short-term fix for this issue, I'm of course prepared to go with a busy-loop over delay. To be sure, could you please research and point out which, if any, of the other functions should be blocking? Thanks!

dok-net, 2019-12-18:
@dirkmueller Did you check, is `readBytes()` the only blocking call inherited by EspSoftwareSerial?

dirkmueller, 2019-12-18:
sorry, was out for the last two days. 

yes, I think it is the only one. the other one I found is readString() which however is not overwritten. 

with #128 being fixed we now have another oddity in the behavior between readBytes in softwareserial and in strream. Stream returns 0 when no bytes are available within timeout, EspSoftwareSerial returns MAX_ULONG (it tries to return -1 but since size_t is unsigned this is being implicitly type converted). 

I'm kind of okay iwth returning a huge value as error condition for calling readBytes() when rx is actually not enabled, but imho when there are no bytes to read it should return 0 rather than super-large value. 

Also, while I totally appreciate the work on Delegate (and I do intend to switch to an event based model) the protocol that I have to parse is "block based" so readBytes(sizeof(mydatastruct)..) is a super useful thing to have..

Regarding blocking, Stream fixes that by handling setTimeout(0) properly (it will then do nonblocking behavior). 

I think to turn off any blocking read setTimeout(0) could be chosen by the user.

dok-net, 2019-12-18:
I've noticed and fixed the (size_t)-1 bug in the line of business :-) Thanks, though.

commit 80c31277 (2019-12-18): Subtle change in API contract: readBytes() are blocking in Stream and HardwareSerial. EspSoftwareSerial adopts this behavior now.

For non-blocking buffer read, use the new read(buffer, size) functions.
Fixes #127
Bump minor release tag.
closed 2019-12-18 by dok-net