plerup/espsoftwareserial #128: espsoftwareserial 6.4.0 readBytes always returns -1

dirkmueller, 2019-12-18:
with the change

https://github.com/plerup/espsoftwareserial/commit/0b9ac7c72ea186180470fbf4096491f376ea4395#diff-67183b415c133795f221633c3e752798L193-R196

The readBytes() method now always returns -1 in 9600-baud-8N1 serial mode, which breaks my application. 

I suggest to do a change similar to this:

```diff
    if (m_parityBuffer && 0 != size) {
        uint32_t parityBits = size;
        while (m_parityOutPos >>= 1) ++parityBits;
        m_parityOutPos = (1 << (parityBits % 8));
        m_parityBuffer->pop_n(nullptr, parityBits / 8);
        return size;
    }
-  return -1;
+   return 0 != size ? size : -1;
```

alternatively returning size always might not be a bad situation either (would work for me), but it is a change in contract/behavior then.

dok-net, 2019-12-18:
OMG - apparently a failure in my testing process. Fixing ASAP.

dok-net, 2019-12-18:
@dirkmueller Please test for regressions and let me know. Thanks!

closed 2019-12-18 by dok-net

dirkmueller, 2019-12-18:
thanks for the quick turnaround. I'll take a look tonight.

dirkmueller, 2019-12-18:
current master seems to work all fine. please release!

commit e1bb39ac (2019-12-18): Fixes #128