plerup/espsoftwareserial #131: Issues on Wemos D1 mini

zarss, 2020-01-02:
Very simple sketch. Works fine on Arduino Nano. On Wemos D1 mini, any value being read returns "4294967295". Sending commands work ok, receiving not.
[J (2).txt](https://github.com/plerup/espsoftwareserial/files/4017311/Jauns.teksta.dokuments.2.txt)

dok-net, 2020-01-02:
Please don't expose us to the dangers of external download links and paste your code here!That said, have you checked if you don't need to call at least begin(9600) on the SW serial object like it's usual direct from your code?Sent from my Huawei Mobile

zarss, 2020-01-02:
Edited.

dok-net, 2020-01-03:
Please check that SoftwareSerial's `begin(...)`is called someplace - otherwise, please ask this question in any support group for the display, this is really no issue with EspSoftwareSerial at all and this is not the place to discuss this. Thank you for your understanding.

closed 2020-01-03 by dok-net