plerup/espsoftwareserial #132: No bytes received while listening

HamzaHajeir, 2020-01-07:
Hello 

I'm facing a problem not receiving bytes using ESP8266 NodeMCU V2 for unexpected times.

I'm using listen() and stopListening() to switch when I want to read a sensor's continuous value. to save MCU's power most of the time.

Some unexpected times, I'm not receiving bytes from the sensor ( available = 0 ) for about 17 seconds (last & longest measured cut). After that it receives without any problem.

That "not receiving" period only comes after calling listen() method.

Here in the image : the right log comes from ESP8266 which listens/stoplistening by a Serial command(any available). (consider it a switch)

At the left is Arduino UNO reading the same Rx pin : It's continuously reading without any interrupt (to make sure Sensor is sending out continuously).

[image]

Example code ( ESP8266 ) : 

constructor + setup : 

```cpp
#define RX 10            //I've added GPIO 10 to isValidGPIOPin function inside SoftwareSerial.cpp
#include <SoftwareSerial.h>

#define BUFFER_SIZE 30
SoftwareSerial ultrasonicSerial(10,250,true,BUFFER_SIZE);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  pinMode(10, INPUT);  //make sure to add this pin in softwareserial isValidGpio funtion
  ultrasonicSerial.begin(9600);
  ultrasonicSerial.enableRx(true);
}
```

in loop() : 
``` cpp
  static uint32_t prevmillis = 0;
  if(active && millis()-prevmillis > 500){
      prevmillis = millis();
      int bytes_available = ultrasonicSerial.available();
      Serial.printf("bytes_available = %d , SWSerial isListening = %d \n", bytes_available, ultrasonicSerial.isListening());

      if (bytes_available >= 16) {
          while (bytes_available / 6.0f > 2.6f) {
              ultrasonicSerial.readStringUntil('R');
              bytes_available = ultrasonicSerial.available();
              Serial.printf("from while loop : bytes_available = %d \n", bytes_available);
          }

          // buffer = ultrasonicSerial.parseInt();
          // if(bytes_available >=6){

          rs232_distance = ultrasonicSerial.parseInt();
          Serial.printf("rs232_distance read = %d\n", rs232_distance);
          ultrasonicSerial.flush();

          // }
      }

  }
```

Where ```active``` is the switched boolean.

Arduino testing code : 

``` cpp

#define RX 10
#include <SoftwareSerial.h>

#define BUFFER_SIZE 30
SoftwareSerial ultrasonicSerial(10, 250, true);
int rs232_distance;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  pinMode(10, INPUT);  //make sure to add this pin in softwareserial isValidGpio funtion
  ultrasonicSerial.begin(9600);

}

void loop() {
  // put your main code here, to run repeatedly:
  static uint32_t prevmillis = 0;
  if (millis() - prevmillis > 500) {
      prevmillis = millis();
      int bytes_available = ultrasonicSerial.available();
      Serial.print("bytes_available = ");
      Serial.println(bytes_available);
      if (bytes_available > 16) {
          while (bytes_available / 6.0f > 2.6f) {
              ultrasonicSerial.readStringUntil('R');
              bytes_available = ultrasonicSerial.available();
              Serial.println("from while loop : bytes_available = "+String(bytes_available));
          }
      }

      bytes_available = ultrasonicSerial.available();
      Serial.println(bytes_available);
  }
}

I'm using : 

ESP8266 core version 2.5.2 (For some compatibility of a framework I'm using).

Any help / suggestion ?

I'm 100% ready to give any info one need for helping.
Thanks.

HamzaHajeir, 2020-01-07:
Just tested out the exact code (just deleted buffer_size parameter) using ESP8266 core v2.6.2 

It seems working well without any wrong behavior.

But for the sake of keeping my code withing one complete esp8266 core, which was 2.5.2. Is there a tweaking in the code/library fixes the issue ?

P.S.  I'm using PlatformIO for this project.

Thanks

HamzaHajeir, 2020-01-23:
Is there a solution ?

dok-net, 2020-01-25:
Backporting is out of the realm of free-of-charge development here, there are a few fixes in the ESP8266 to support later versions of EspSoftwareSerial.

closed 2020-01-25 by dok-net

HamzaHajeir, 2020-01-25:
> Backporting is out of the realm of free-of-charge development here, there are a few changes in the ESP8266 core to support later versions of EspSoftwareSerial.

Right, I've just tried to back-port, It's too hard as there's a lot of roots changed along in the core.

Anyway I've resolved it by updating the core, It's now under-test. as framework paused supporting at 2.5.0 .