plerup/espsoftwareserial #133: Cache disabled but cached memory region accessed

executer-uno, 2020-01-13:
Hi,
Not sure the root reason, but I time to time got such panic error. Im using ESP32 with
HW USART x2 ports (RX + TX)
SW USART x1 port (RX + TX)
WiFi data transfer to google spreadsheets by https redirect lib
SQLite ESP library on MicroSD card
And I2C OLED display

I can easyly call that error by sending 200+ bytes of data to SW serial (buffer setted to 1024 bytes)
I also can catch error on less data amount, looks like it is just interference between some librarys, more time SW Serial busy - higher chanches to panic.
`		serialGPS.begin(9600, GPS_SERIAL_RX, GPS_SERIAL_TX, SWSERIAL_8N1, false, 1024 );`

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x401e1d48  PS      : 0x00060034  A0      : 0x8008128c  A1      : 0x3ffbe780  
A2      : 0x3ffc2854  A3      : 0xaa5eeebb  A4      : 0x3ffc3694  A5      : 0x00000000  
A6      : 0x3ffc3690  A7      : 0xffffffff  A8      : 0x8008108d  A9      : 0x00000001  
A10     : 0x3ffd0114  A11     : 0x3ffbe78c  A12     : 0x3ffba264  A13     : 0x0000abab  
A14     : 0x3ffc5be0  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40087924  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x401e1d48

Backtrace: 0x401e1d48:0x3ffbe780 0x40081289:0x3ffbe7b0 0x40085669:0x3ffbe7d0 0x40087921:0x3ffba190 0x40083a63:0x3ffba1b0 0x40089a75:0x3ffba1d0
```

I also have a JTAG debugger, but not quite clear how to find the problem's root.
```
PM25_portable.elf	
	Thread #13 1073447000 (Name: loopTask) (Suspended : Container)	
		String::wbuffer() at WString.h:282 0x40140385	
		String::copy() at WString.cpp:213 0x40140385	
		String::String() at WString.cpp:35 0x401403ee	
		loop() at PM25_portable.ino:4,934 0x400f4085	
		loopTask() at main.cpp:19 0x40144a64	
		vPortTaskWrapper() at port.c:143 0x40089a78	
	Thread #14 1073465444 (Name: IDLE1) (Suspended : Container)	
		esp_pm_impl_waiti() at pm_esp32.c:492 0x401e2136	
		esp_vApplicationIdleHook() at freertos_hooks.c:63 0x40147cc1	
		prvIdleTask() at tasks.c:3,382 0x4008b26c	
		vPortTaskWrapper() at port.c:143 0x40089a78	
	Thread #15 1073464032 (Name: IDLE0) (Suspended : Container)	
		esp_pm_impl_waiti() at pm_esp32.c:492 0x401e2136	
		esp_vApplicationIdleHook() at freertos_hooks.c:63 0x40147cc1	
		prvIdleTask() at tasks.c:3,382 0x4008b26c	
		vPortTaskWrapper() at port.c:143 0x40089a78	
	Thread #16 1073468136 (Name: Tmr Svc) (Suspended : Container)	
		prvProcessTimerOrBlockTask() at timers.c:589 0x4008bf90	
		prvTimerTask() at timers.c:544 0x4008c08f	
		vPortTaskWrapper() at port.c:143 0x40089a78	
	Thread #17 1073455716 (Name: ipc1, State: Running @CPU1) (Suspended : Signal : SIGTRAP:Trace/breakpoint trap)	
		circular_queue<unsigned int>::push() at circular_queue.h:268 0x401e1d48	
		__onPinInterrupt() at esp32-hal-gpio.c:218 0x4008128c	
		_xt_lowint1() at xtensa_vectors.S:1,154 0x4008566c	
	Thread #18 1073445320 (Name: esp_timer) (Suspended : Container)	
		esp_crosscore_int_send_yield() at crosscore_int.c:112 0x40082fda	
		xQueueGenericReceive() at queue.c:1,592 0x4008988c	
		timer_task() at esp_timer.c:316 0x40148b47	
		vPortTaskWrapper() at port.c:143 0x40089a78	
	Thread #19 1073446424 (Name: ipc0) (Suspended : Container)	
		xQueueGenericReceive() at queue.c:1,592 0x4008988c	
		ipc_task() at ipc.c:51 0x40083a33	
		vPortTaskWrapper() at port.c:143 0x40089a78	
	Thread #21 1073438112 (Name: wifi, State: Running @CPU0) (Suspended : Container)	
		panicHandler() at panic.c:251 0x4008d7ba	
		xt_highint4() at dport_panic_highint_hdl.S:117 0x400838fd	
		0x40040025	
		0x400621d5	
		esp_rom_spiflash_wait_idle() at spi_flash_rom_patch.c:35 0x40093495	
		esp_rom_spiflash_program_page_internal() at spi_flash_rom_patch.c:191 0x400937eb	
		esp_rom_spiflash_write() at spi_flash_rom_patch.c:476 0x40093a40	
		spi_flash_write_inner() at flash_ops.c:266 0x40087c30	
		spi_flash_write() at flash_ops.c:397 0x40087ea7	
		nvs::nvs_flash_write() at nvs_ops.cpp:66 0x4017e858	
		<...more frames...>	
	Thread #22 1073437100 (Name: tiT) (Suspended : Container)	
		esp_crosscore_int_send_yield() at crosscore_int.c:112 0x40082fda	
		xQueueGenericReceive() at queue.c:1,592 0x4008988c	
		sys_arch_mbox_fetch() at sys_arch.c:297 0x401c91c8	
		sys_timeouts_mbox_fetch() at timeouts.c:430 0x401b9959	
		tcpip_thread() at tcpip.c:109 0x401b23d3	
		vPortTaskWrapper() at port.c:143 0x40089a78	
	Thread #23 1073428176 (Name: network_event) (Suspended : Container)	
		xQueueGenericReceive() at queue.c:1,592 0x4008988c	
		semphr_take_wrapper() at esp_adapter.c:272 0x40146d08	
		ieee80211_ioctl() at 0x4018e0c5	
		esp_wifi_set_config() at 0x4018327a	
		WiFiSTAClass::begin() at WiFiSTA.cpp:204 0x400f8b72	
		WiFiGenericClass::_eventCallback() at WiFiGeneric.cpp:371 0x400f8324	
		_network_event_task() at WiFiGeneric.cpp:61 0x400f8564	
		vPortTaskWrapper() at port.c:143 0x40089a78	
	Thread #24 1073432756 (Name: eventTask) (Suspended : Container)	
		xQueueGenericReceive() at queue.c:1,592 0x4008988c	
		esp_event_loop_task() at event_loop.c:53 0x40148878	
		vPortTaskWrapper() at port.c:143 0x40089a78	
	Thread #25 1073591444 (Name: mdns) (Suspended : Container)	
		xQueueGenericReceive() at queue.c:1,592 0x4008988c	
		_mdns_service_task() at mdns.c:3,988 0x4017a37a	
		vPortTaskWrapper() at port.c:143 0x40089a78	
	Thread #26 2 (Name: ipc1, Interrupted @CPU1) (Suspended : Container)	
		spi_flash_op_block_func() at cache_utils.c:82 0x40087919	
		ipc_task() at ipc.c:62 0x40083a66	
		vPortTaskWrapper() at port.c:143 0x40089a78	
```
(trapped on Thread #17 1073455716 (Name: ipc1, State: Running @CPU1) (Suspended : Signal : SIGTRAP:Trace/breakpoint trap)) see scrinshot - I was trying different ports and it always traps on push() function in SW Serial RX pin interrupt (pin=16 on screenshot)
[image]

Probably any one have Idea what is the reason?

dok-net, 2020-01-25:
@executer-uno I can't quite follow your conclusion from the stack traces, my reading of it is that probably there is a problem with IRQ handling and not-in-IRAM-cache code, thread 21. Why did you conclude that the EspSoftwareSerial thread is involved?

dok-net, 2020-02-26:
Closing due to no further feedback activity.

closed 2020-02-26 by dok-net